fzi-forschungszentrum-informatik/gpu-voxels #5: About Display Problems(./octree_provider )?

NAMCER, 2015-03-12:
HI!
I follow your advice:
1.Modification /home/workspace /gpu-voxels-master/packages/gpu-voxels/src/examples/Config_live_col_detect.xml
===================================================
  <camera>
  <position>
    <x> 8314.02 </x>
    <y> 8697.94 </y>
    <z> 8211.91 </z>
  </position>
  <horizontal_angle> 341.519 </horizontal_angle> 
  <vertical_angle> 0.264989 </vertical_angle> 
  <field_of_view> 60 </field_of_view> 
</camera>
===================================================
2.Running examples command:
./octree_provider -shm 0 -m kinect_live -resTree 20 -resFree 20 -resOcc 20 -mem 1024 -fps 30 -c -shm 1 -m load_pcd -f ./robot4cmRes.pcd -resTree 20 -resFree 20 -resOcc 20 -x 20 -y 180 -z -10

×The results：
1.I did not see anything voxle screen.
2. My Terminal:

Squelsh, 2015-03-12:
You have to launch two programs: First the provider (as you wrote above) and after that the visualizer with the configfile that you changed.
What do you see then? An empty window? What is the output of the provider on the console?

NAMCER, 2015-03-15:
I reinstall gpu-voxels
cmake .. -DCMAKE_BUILD_TYPE=Release
.
.
.
-- ---- Scanning packages -------------------------------------------------------------
-- 
-- Info:    icl_core -- building library without support for [OpenSpliceDDS].
-- Info:    icl_core_config -- building library without support for [OpenSpliceDDS].
-- Info:    icl_core_logging -- building library without support for [Eigen2].
-- Found ROS: ... yes
-- Found LibRt: ... yes
CMake Warning at packages/gpu_voxels/src/gpu_visualization/CMakeLists.txt:16 (FIND_PACKAGE):
  Could not find module FindGLEW.cmake or a configuration file for package
  GLEW.

  Adjust CMAKE_MODULE_PATH to find FindGLEW.cmake or set GLEW_DIR to the
  directory containing a CMake configuration file for GLEW.  The file will
  have one of the following names:

    GLEWConfig.cmake
    glew-config.cmake

-- Could NOT find GLM (missing:  GLM_INCLUDE_DIR) 
-- Warning: gpu_voxels_visualization_core -- not building library, missing [GLEW;GLM].
-- Warning: gpu_voxels_visualizer -- not building executable, missing [gpu_voxels_visualization_core].
-- Warning: ibeo_source -- not building executable, missing   [icl_sourcesink;icl_hardware_ibeo_noapi;icl_hardware_ncom;icl_gps_types].
-- Configuring done
-- Generating done
.
.
.
I do not know if it is because of these two  (missing:  GLM_INCLUDE_DIR ,and ,FindGLEW.cmake) ?

NAMCER, 2015-03-15:
But make commaned not error
bin/
icl_core_performance_monitor_example  swept_volume_vs_environment  ts_icl_core_config
octree_bench                          test_gpu_voxels_core         ts_icl_core_crypt
octree_provider                       test_icl_core_config         ts_icl_core_thread
octree_test                           test_icl_core_convert        uls
robot_vs_environment                  test_icl_core_logging
sandbox                               ts_icl_core

Squelsh, 2015-03-15:
Dear Namcer,

no offense, but this is a bugtracker for the GPU-Voxels library and not a forum to learn about CMake. Please try to find answers to such basic stuff by a Google search or CMake tutorials.
Make does not end in an error, as CMake will not even start compiling with make, as long as not all dependencies are found.
Install the required libs (find a list on http://gpu-voxels.org/documentation/prerequisites/ ) and try again.

I will close this issue.

Best,
Andreas

closed 2015-03-15 by Squelsh

praveenneuron, 2015-06-05:
Follow this if 

download zip from
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels
##
echo "export GPU_VOXELS_MODEL_PATH=/home/aa/gpu-voxels-master/packages/gpu_voxels/models" >> ~/.bashrc
#install required libraries  http://www.gpu-voxels.org/documentation/prerequisites/
##pcl
sudo add-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl
sudo apt-get update
sudo apt-get install libpcl-all
#others
sudo apt-get install libglew1.10
sudo apt-get install libglew-dev
sudo apt-get install libglew-dbg
sudo apt-get install libglm-dev 
sudo apt-get install libcppunit-1.13-0
sudo apt-get install libcppunit-dev
sudo apt-get install doxygen

##extract at home
cd ~/gpu-voxels-master/
mkdir build
cd build
cmake ..
make
#if you want to use it in some other project use install option.... files will be saved in export folder
make install