fzi-forschungszentrum-informatik/gpu-voxels #6: How should I run visualizer "gpu_voxels_visualization -c configfile.xml"

praveenneuron, 2015-03-13:
After compiling succefully I am only able to run command line examples. How can I run visualizer. Also can you provide tutorial to run with ROS. That will be of great help to experiment.

Squelsh, 2015-03-13:
The viewer connects to the provider via shared memory. Just open another console and start the viewer with `gpu_voxels_visualization -c configfile.xml`.

How would a ROS example help you? It is exactly the same API as in the example files. Just link against GPU Voxels with your ROS Node and you can use the API. You yourself will than have to take care on how to interact with the lib via ROS messages, services whatsoever.

Apart from that I am currentyl working on a URDF Parser for GPU Voxels, so that you can load ROS kinematics descriptions easily.

closed 2015-03-13 by Squelsh

praveenneuron, 2015-03-14:
URDF Parser for GPU Voxels would help. Can I know by when are you releasing it?

Squelsh, 2015-03-14:
Just sign up for the mailing list and you will allways be informed about all updates.
Right now I can't tell you how long it will take to implement the stuff. Hopefully 1-2 weeks, depending on my workload.