fzi-forschungszentrum-informatik/gpu-voxels #12: I want to external project using gpu-voxels

Hooney1011, 2015-05-11:
Hi 
Now, I am compiling the external project using gpu-voxels.
I have some error.
I wrote in mine CMakeLists.txt 
INCLUDE_DIRECTORIES(${gpu_voxels_INCLUDE_DIRS}  ${icl_core_INCLUDE_DIRS}    ${tinyxml_INCLUDE_DIRS})
 FIND_PACKAGE(gpu_voxels REQUIRED )
 TARGET_LINK_LIBRARIES(MY_TARGET_NAME ${gpu_voxels_LIBRARIES})

I used 
include icl_core_config/Config.h
include gpu_voxels/GpuVoxels.h  
include gpu_voxels/helpers/Kinect.h
include gpu_voxels/helpers/MetaPointCloud.h
include gpu_voxels/helpers/PointcloudFileHeandler.h
include gpu_voxels/logging/logging_gpu_voxels.h
 in mine code.
ERROR:
In file included from /usr/local/include/icl_core/os_fs.h:30:0,
                 from /usr/local/include/icl_core/os.h:30,
                 from /usr/local/include/icl_core/TimeBase.h:36,
                 from /usr/local/include/icl_core/TimeSpan.h:31,
                 from /usr/local/include/icl_core_logging/Logging.h:32,
                 from /usr/local/include/icl_core_config/Config.h:37,
                 from /home/slam/workspace/MYproject/viewer.cpp:23:
/usr/local/include/icl_core/os_ns.h:34:3: error: #error "No os namespace implementation defined for this platform."
 # error "No os namespace implementation defined for this platform."

/usr/local/include/icl_core/os_fs.h: In function ‘int icl_core::os::rename(const char*, const char*)’:
/usr/local/include/icl_core/os_fs.h:50:10: error: ‘ICL_CORE_OS_IMPL_NS’ has not been declared return ICL_CORE_OS_IMPL_NS::rename(old_filename, new_filename);
.....
What should i do ?

Squelsh, 2015-05-12:
Hello!
Are you LINANZHE from the mailing list? Or is it by chance, that he also tries to link against GPU-Voxels?

So you use CMake also for you external project.
Main point: I think you missed to set the definitions for icl_core that define some stuff (see below).
Your OS is some kind of Linux? Seems that _SYSTEM_POSIX_ is not defined. It should be, after using the definitions you find below.

HTH,
Andreas

```
Example CMakeLists.txt:
-------------------------------

find_package(gpu_voxels)
find_package(icl_core)

add_definitions(
  ${gpu_voxels_DEFINITIONS}
  ${icl_core_DEFINITIONS}
)

include_directories(
   ${CUDA_INCLUDE_DIRS}
   ${Boost_INCLUDE_DIRS}
   ${PCL_INCLUDE_DIRS}
   ${gpu_voxels_INCLUDE_DIRS}
   ${icl_core_INCLUDE_DIRS}
)

target_link_libraries(ros_gpu_voxels_node
     ${icl_core_LIBRARIES}
     ${gpu_voxels_LIBRARIES}
     ${CUDA_LIBRARIES}
)
```

Squelsh, 2015-11-10:
Closed due to inactivity.

closed 2015-11-10 by Squelsh