fzi-forschungszentrum-informatik/gpu-voxels #13: error -  std::bad_alloc: an illegal memory access was encountered

NAMCER, 2015-05-21:
I am using Kinect sensor big point cloud data.
when I launch visualized gpu-voxels .

ERROR:
==========================================================================
<2015-05-21 22:59:07.931> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home//workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/robot/KinematicChain.cu on line 122.
<2015-05-21 22:59:07.931> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/robot/KinematicChain.cu on line 125.
<2015-05-21 22:59:07.944> Gpu_voxels_helpers(Warning) MetaPointCloud::updatePointCloud: Size of pointcloud changed! Rearanging memory
<2015-05-21 22:59:07.944> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/MetaPointCloud.cu on line 218.
<2015-05-21 22:59:07.944> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/MetaPointCloud.cu on line 220.
<2015-05-21 22:59:07.944> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/MetaPointCloud.cu on line 222.
<2015-05-21 22:59:07.944> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/MetaPointCloud.cu on line 224.
<2015-05-21 22:59:07.974> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/MetaPointCloud.cu on line 55.
<2015-05-21 22:59:07.974> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/MetaPointCloud.cu on line 58.
<2015-05-21 22:59:07.974> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/MetaPointCloud.cu on line 62.
<2015-05-21 22:59:07.974> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/MetaPointCloud.cu on line 66.
<2015-05-21 22:59:07.974> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/MetaPointCloud.cu on line 70.
<2015-05-21 22:59:07.974> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/MetaPointCloud.cu on line 83.
<2015-05-21 22:59:07.974> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/MetaPointCloud.cu on line 95.
<2015-05-21 22:59:07.974> Gpu_voxels_helpers(Info) MetaPointCloud::init: This MetaPointCloud requires: 63.430595MB on the GPU and on the Host
<2015-05-21 22:59:07.993> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/MetaPointCloud.cu on line 267.
<2015-05-21 22:59:07.994> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/GvlNTree.hpp on line 184.
<2015-05-21 22:59:07.994> CudaLog(Error)::cuHandleError: an illegal memory access was encountered in /home/workspace/multikinect-voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/GvlNTree.hpp on line 185.
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'thrust::system::detail::bad_alloc'
  what():  std::bad_alloc: an illegal memory access was encountered
Aborted (core dumped)
==========================================================================
I don't know why this is happened.
If you know ,Please tell me .
Thank you very much.

Squelsh, 2015-05-22:
Too little information... Could be everything or nothing.
What are you running, which hardware, are the example programs that use a Kinect are running correctly on your setup?
Perhaps you do not have enough RAM on your GPU? Or a wrong transformation for the Kinect and so you are trying to insert data outside of the Octree?

Squelsh, 2015-11-10:
Closed due to inactivity.

closed 2015-11-10 by Squelsh