fzi-forschungszentrum-informatik/gpu-voxels #15: error while compile

baishibona, 2015-06-05:
Hi,

While compiling with "make" under "build" folder, I was having the following errors which stops at ~90%

/home/cuda/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/Octree.cu(57): here

/home/cuda/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp(1874): error: namespace "thrust" has no member "minmax_element"

/home/cuda/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp(1876): error: namespace "thrust" has no member "minmax_element"

/home/cuda/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp(1878): error: namespace "thrust" has no member "minmax_element"

/home/cuda/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp(3044): error: namespace "thrust" has no member "minmax_element"

/home/cuda/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp(3046): error: namespace "thrust" has no member "minmax_element"

/home/cuda/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp(3048): error: namespace "thrust" has no member "minmax_element"

Would you please indicate what might cause the problem?
I am using Ubuntu 14.04 without ros installed.
Thanks a lot!

Squelsh, 2015-06-07:
Hi Baishibona,

which version of CUDA do you use?
The minmax_element should be there: http://docs.thrust.googlecode.com/hg/group__extrema.html
Did you install Thrust via aptitude?

Best,
Andreas

baishibona, 2015-06-07:
Hi Andreas,

I am using CUDA 7.0 and Ubuntu 14.04LTS. 
I thought Thrust comes within CUDA, should I install it as a prerequisite? 
Any other package I might missing other than "CUDA, PCL, OPENNI, BOOST"?
Thanks for the help!

./Bona

Squelsh, 2015-06-07:
Hi Bona!

You are right about Thrust. Nevertheless there is a separate independent package under Ubuntu (which I do not use). So my first thought was, that you installed that one.

So, I think you should try it with CUDA 6.5. I never built with CUDA 7 up to now, as this made problems with my Ubuntu 15.04 (boost problems).

I will have a look at the Thrust incompatibility as soon as I can switch to 7.0.

Good luck, and please report back, if it works with 6.5.

Andreas

baishibona, 2015-06-07:
Thanks a lot Andreas!
Yes definitely I will keep you updated. 

./Bona

Squelsh, 2015-06-08:
Hi Bona,

I found the bug: '''#include <thrust/extrema.h>''' simply was missing, and in older versions of Thrust some include dependency included it anyhow.
So with the next release (coming today or tomorrow) this will be fixed.
But I still have to check, if everything else works with CUDA 7.

Thanks for pointing out the error,
Andreas

baishibona, 2015-06-08:
> I found the bug: '''#include ''' simply was missing, and in older versions

Thanks Andreas!
I guess I will reinstall CUDA 6.5 and try the new release today or tomorrow.

./Bona

baishibona, 2015-06-10:
Hi Andreas,

I got the gpu-voxel compiled successfully with CUDA 6.5 (however not working with CUDA7), even without openni(I installed it but gpu-voxel failed to locate it).
Thanks for the help!

./Bona

Squelsh, 2015-06-10:
Hi Bona,

do you mean that CMake can't find OpenNi? Or is it the driver that can't 
find the Kinect during runtime?
In the first case, the FindScript could be missleading and compilation 
could still succeed.
In the second case you could switch to the old OpenNi that supports 
Kinect (OpenNi2 only supports Asus Xtion and some other cams but not the 
Kinect).
Or you could use any driver/cam combo you prefer, as long as it delivers 
a poincloud (:

Good luck,
Andreas

baishibona, 2015-06-11:
The FindScript failed to locate openni. However I did successfully compiled
it.
Thanks!

baishibona, 2015-06-11:
Hi Andreas,

Sorry I missed your second question. The problem with CUDA 7 is compile issue, errors occurred at about 95% of compile succeed.

./Bona

Squelsh, 2015-06-15:
Hi!
I fixed the OpenNi find problem. The report at the beginning of the CMake run should be fine now.

You don't have the exact error from Cuda 7.0, do you? Currently I can not reproduce the compilation, as with Ubuntu 15.04 a CUDA has a bug with the latest Boost 1.55 lib ):
I opened a new issue ( https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/issues/17 ) for that toppic and will close this one now. If you can reproduce the error, please post it there.

Cheers, Andreas

closed 2015-06-15 by Squelsh