fzi-forschungszentrum-informatik/gpu-voxels #17: GPU Voxels does not compile with CUDA 7.0

Squelsh, 2015-06-15:
Errors occur at about 95% of compilation.
Details will follow.

baishibona, 2015-08-17:
note: ‘bool gpu_voxels::NTree::isValidValue(gpu_voxels::NTree::Environment::NodeProb::RayCastType)’ declared here, later in the translation unit
 bool isValidValue(const Environment::NodeProb::RayCastType a)
 ^
CMake Error at gpu_voxels_octree_CUDA_TARGET_generated_Octree.cu.o.cmake:264 (message):
  Error generating file

Squelsh, 2015-08-18:
I think, I will not target CUDA 7.0 as a supported platform, as there are problems between recent Boost Versions and CUDA 7.0.
See bug described here: https://svn.boost.org/trac/boost/ticket/11391#comment:6
So currently I am planning to directly switch to CUDA 7.5 for the 1.0 release of GPU Voxels.
Till then, I will stick to CUDA 6.5 and rather include more features and fix bugs.

baishibona, 2015-08-18:
Hi Andreas,

Thanks for the explanation. I am planning to switch back to CUDA 6.5 as gpu
voxel is the main reason I install CUDA.
Thanks again and looking forward to 1.0 release!

./Bona

Squelsh, 2015-11-10:
This issue is fixed with the new 0.8 release. It complies to CUDA 7.5 now.

closed 2015-11-10 by Squelsh