fzi-forschungszentrum-informatik/gpu-voxels #36: Cannot Compile with CUDA 7.5

Maverobot, 2016-08-25:
I followed the documentation to compile the project. However, after trying to "cmake ..", an error occured.

`CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108 (message):
  Could NOT find CUDA: Found unsuitable version "7.5", but required is exact
  version "5.5" (found /usr/local/cuda)`

Does it not support CUDA 7.5 or is there something that I missed?

Thanks for your time.

Squelsh, 2016-08-25:
Hi!
This is a problem with some other FindScript (perhaps OpenCV), which requires a special version.
Try to run cmake with the option -DICMAKER_USE_PERCEPTION_PACKAGES=false
This should suppress the search for OpenCV.
Please report, if this helped.

Maverobot, 2016-08-25:
It works like a charm. Thanks!

closed 2016-08-26 by Maverobot