fzi-forschungszentrum-informatik/gpu-voxels #37: No os namespace implementation defined for this platform

Maverobot, 2016-09-02:
Platform: Ubuntu 14.04 64bit
IDE: Qtcreator, QT4

I am trying to create my own QT project which uses Gpu-voxels as library. But it fails in compilation and gives quite a few errors like this: 

`/home/include/icl_core/os_mem.h:36: error: #error "No os_mem implementation defined for this platform."
 # error "No os_mem implementation defined for this platform."`

Am I missing something here?

Thanks for your time.

Squelsh, 2016-09-02:
Uh, never tried that...
But from the error that you posted, I assume that some defines are missing, that should have been set by CMake.
Our ICL Libs require a define, that tells them about the running architecture and OS.

Could you give me some hints on how you created the Qt project so that I can try to reproduce the error?

Maverobot, 2016-09-02:
Thanks for the quick reply.

What I did was just copying the Sandbox.cpp from your examples folder into
a new folder. Then I created a standard .pro file and add the installed
"include" folder in the include path.

Squelsh, 2016-09-02:
Hmm. I don't know, what QtCreator does for you in that case, but the problem you experience is not a GPU-Voxels specific thing (:
I don't guarantee that I exported all dependencies in a correct way for a "make install", but what you see is a CMake dependency problem.

What I normally do when I link against a lib that uses CMake is:
Build the lib, install it into an export directory (defined via `CMAKE_INSTALL_PREFIX`).

Then I create my project and add the lib as a dependency in my own `CMakeLists.txt` file.
At the first `cmake` run, I add the export directory to the `CMAKE_PREFIX_PATH`.

Then `cmake` should find the library and automatically inherit the required include- and lib-paths.

How experienced are you with CMake?

Maverobot, 2016-09-02:
Thanks for the suggestion. I have been using cmake for quite a long time.
But I have to use qt for integration in another project.

I will look into this a bit more, and will get you informed if any solution
is found.

Squelsh, 2016-09-02:
Well, Qt Creator uses CMake to deal with all dependencies. So it should be no real problem. Nevertheless I don't know the difference between the .pro-File and a new CMakeLists.

I had a look at you reported error message, and I was right: The precompiler is missing either
`_SYSTEM_POSIX_` or `_SYSTEM_WIN32_` to be defined. Which is normally set via our IcMaker build system which is a convenience-add-on on top of CMake that is run, when you execute `cmake`.

Good luck!

Squelsh, 2016-09-02:
BTW: You are from TUM? Are you involved in http://www.horse-project.eu/ ?

Maverobot, 2016-09-05:
Really nice hint there. Thank you! I got it compiled and running in QtCreator by adding the following line in .pro file

`DEFINES += "_SYSTEM_POSIX_=_SYSTEM_ LINUX_"`

BTW, I am not a student from TUM anymore and not involved in the [horse-project](http://www.horse-project.eu/).  :D

Squelsh, 2016-09-06:
Happy to hear that you could make it compile.
I will somewhere put a hint for others about those two problems that you encountered and close this issue.

PS: I am in the HORSE project together with TUM, thats why I asked.
Tell me about your project and how you want to use GPU Voxels (via mail), if you find some time. I am happy to share my experience.

closed 2016-09-06 by Squelsh