fzi-forschungszentrum-informatik/gpu-voxels #38: fabs() / sqrt() was not declared in this scope

Maverobot, 2016-09-06:
When I try to compile my own project using GPU-voxels as a library, errors occur in `gpu_voxles/helpers/cuda_datatypes.h`:

`error: 'fabs' was not declared in this scope`
`error: 'sqrt' was not declared in this scope`

Adding the header `#include <math.h>` in `cuda_datatypes.h` could solve this.

Squelsh, 2016-09-07:
What exactly do you include in your project?
The toplevel `gpu_voxels`? Which headers do you include in your project?
Which compiler throws the error? nvcc or gcc?

Squelsh, 2016-09-14:
Could you solve the problem?

BTW: I just added an example on how to link against GPU-Voxels via plain CMake.

Squelsh, 2016-09-20:
Please try the example https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/tree/master/example_how_to_link of how to link against GPU-Voxels.
If this does not help, reopen this issue. Thanks.

Closed due to inactivity.

closed 2016-09-20 by Squelsh