fzi-forschungszentrum-informatik/gpu-voxels #44: collideWithTypes returns empty types_in_collision for BitVectorVoxelList <> (ProbVoxelMap and BitVectorVoxelMap)

wxmerkt, 2017-01-10:
Hi,
Thank you very much for open sourcing this library. I've been using both the 9.1 and now the 9.3 Xmas Edition release. I create an environment as well as a swept volume and check for collisions with collideWithTypes. I've benchmarked different types for both the environment as well as the swept volume and noticed that not all types return the correct/same result for the same query, i.e [the swept volume is always the latter]:

1. ``MT_BITVECTOR_VOXELLIST collideWithTypes MT_PROB_VOXELMAP``: reports a valid number of collisions (60-70 in my test case), but the ``types_with_collision`` BitVectorVoxel has all voxels marked as "0"
2. The same goes for ``MT_BITVECTOR_VOXELLIST collideWithTypes MT_BITVECTOR_VOXELMAP`` (i.e. the types aren't returned)
3. ``MT_BITVECTOR_OCTREE collideWithTypes MT_BITVECTOR_VOXELMAP`` and ``MT_BITVECTOR_OCTREE collideWithTypes MT_BITVECTOR_VOXELLIST`` work fine and return both ``num_cols`` as well as ``types_in_collision`` that are valid.

In all examples the code to generate the swept volume and assign the bit vector meaning follows the example and is unchanged - only the types of the maps is changed.

Are these collision pairs supposed to work and is there something I might be missing?

Thank you very much,
Wolfgang

Squelsh, 2017-01-10:
Hi Wolfgang,
thanks for your appreciation and sorry, if you hit a Bug!
I will investigate the functions that you listed, set up Boost-Tests-Cases for them and report the results here.

In the meantime it would be nice if you could run the Boost Test Suite (2 times) and send me the generated report. The executable is called: `test_gpu_voxels_core`. You have to set the `GPU_VOXELS_MODEL_PATH` first by executing: `export GPU_VOXELS_MODEL_PATH=/my/home/src/ic_workspace_gpu/packages/gpu_voxels/models/`

Cheers,
Andreas

wxmerkt, 2017-01-10:
Hi Andreas,
Thanks for the quick response -  [here](https://gist.github.com/wxmerkt/b46b7f9c3fbe9048a529451345716668)'s the output of the test suite.

Thank you very much for looking into this, cheers,
Wolfgang

Squelsh, 2017-01-10:
Hm, so for your first point in the list I can not confirm your results.
But I have to admit that the API is not that intuitive. The Voxel that you get as a result is only a single Voxel, whose BitVector contains the Types that lie in collision... 

See the testcase that I put together:

```
BOOST_AUTO_TEST_CASE(bitvoxellist_collide_with_types_prob_voxelmap)
{
  PERF_MON_START("bitvoxellist_collide_with_types_prob_voxelmap");
  for(int i = 0; i < iterationCount; i++)
  {
    float side_length = 1.f;

    BitVectorVoxelList* list = new BitVectorVoxelList(Vector3ui(dimX, dimY, dimZ), side_length, MT_BITVECTOR_VOXELLIST);

    Vector3f b1_min(1.1,1.1,1.1);
    Vector3f b1_max(3.9,3.9,3.9);
    Vector3f b2_min(2.1,2.1,2.1);
    Vector3f b2_max(4.9,4.9,4.9);

    std::vector<BitVoxelMeaning> voxel_meanings;
    voxel_meanings.push_back(BitVoxelMeaning(11));
    voxel_meanings.push_back(BitVoxelMeaning(12));

    std::vector<std::vector<Vector3f> > box_clouds;
    float delta = 0.1;

    box_clouds.push_back(createBoxOfPoints(b1_min, b1_max, delta));
    box_clouds.push_back(createBoxOfPoints(b2_min, b2_max, delta));

    MetaPointCloud boxes(box_clouds);
    boxes.syncToDevice();

    list->insertMetaPointCloud(boxes, voxel_meanings);

    GpuVoxelsMapSharedPtr map_2(new ProbVoxelMap(Vector3ui(dimX, dimY, dimZ), side_length, MT_PROBAB_VOXELMAP));
    map_2->insertMetaPointCloud(boxes, eBVM_OCCUPIED);

    BitVectorVoxel types_in_collision;

    size_t num_colls = list->collideWithTypes(map_2->as<ProbVoxelMap>(), types_in_collision, 1.0);
    BOOST_CHECK_MESSAGE(num_colls == 46, "Number of Collisions == 46");
    BOOST_CHECK_MESSAGE(types_in_collision.bitVector().getBit(11) && types_in_collision.bitVector().getBit(12), "Both Types found.");
    types_in_collision.bitVector().clearBit(11);
    types_in_collision.bitVector().clearBit(12);
    BOOST_CHECK_MESSAGE(types_in_collision.bitVector().isZero(), "All other Types are clear.");
    PERF_MON_SILENT_MEASURE_AND_RESET_INFO_P("bitvoxellist_collide_with_types_prob_voxelmap", "bitvoxellist_collide_with_types_prob_voxelmap", "voxellists");
  }
}
```

Squelsh, 2017-01-11:
Hi again,

I took the same testcode and applied it to your second case. It works for me.
So perhaps I did not understand your problem or what your expected outcome is...

BTW: The results from the test protocol are perfect. The only error is a "Todo" for the next release that I missed to remove from the release code...

So here is the test code. 

`BOOST_AUTO_TEST_CASE(bitvoxellist_collide_with_types_bitvoxelmap)
{
  PERF_MON_START("bitvoxellist_collide_with_types_bitvoxelmap");
  for(int i = 0; i < iterationCount; i++)
  {
    float side_length = 1.f;

    BitVectorVoxelList* list = new BitVectorVoxelList(Vector3ui(dimX, dimY, dimZ), side_length, MT_BITVECTOR_VOXELLIST);

    Vector3f b1_min(1.1,1.1,1.1);
    Vector3f b1_max(3.9,3.9,3.9);
    Vector3f b2_min(2.1,2.1,2.1);
    Vector3f b2_max(4.9,4.9,4.9);

    std::vector<BitVoxelMeaning> voxel_meanings;
    voxel_meanings.push_back(BitVoxelMeaning(11));
    voxel_meanings.push_back(BitVoxelMeaning(12));

    std::vector<std::vector<Vector3f> > box_clouds;
    float delta = 0.1;

    box_clouds.push_back(createBoxOfPoints(b1_min, b1_max, delta));
    box_clouds.push_back(createBoxOfPoints(b2_min, b2_max, delta));

    MetaPointCloud boxes(box_clouds);
    boxes.syncToDevice();

    list->insertMetaPointCloud(boxes, voxel_meanings);

    GpuVoxelsMapSharedPtr map_2(new BitVectorVoxelMap(Vector3ui(dimX, dimY, dimZ), side_length, MT_BITVECTOR_VOXELMAP));
    map_2->insertMetaPointCloud(boxes, eBVM_OCCUPIED);

    BitVectorVoxel types_in_collision;

    size_t num_colls = list->collideWithTypes(map_2->as<BitVectorVoxelMap>(), types_in_collision, 1.0);
    BOOST_CHECK_MESSAGE(num_colls == 46, "Number of Collisions == 46");
    BOOST_CHECK_MESSAGE(types_in_collision.bitVector().getBit(11) && types_in_collision.bitVector().getBit(12), "Both Types found.");
    types_in_collision.bitVector().clearBit(11);
    types_in_collision.bitVector().clearBit(12);
    BOOST_CHECK_MESSAGE(types_in_collision.bitVector().isZero(), "All other Types are clear.");
    PERF_MON_SILENT_MEASURE_AND_RESET_INFO_P("bitvoxellist_collide_with_types_bitvoxelmap", "bitvoxellist_collide_with_types_bitvoxelmap", "voxellists");
  }
}`

Cheers,
Andreas

wxmerkt, 2017-01-11:
Thanks! I will have another look - might there be an issue that I am using ``insertBoxIntoMap`` that could be treated differently? What are general swept-volume-generation and collision-check times you're seeing on your hardware? We are using a GTX 970 for a 1.2 x 1.2 x 2 volume at 5cm voxel size and are seeing ~20ms.

Cheers,
Wolfgang

Squelsh, 2017-01-24:
Hi Wolfgang,
could you find the problem with the collision Bit-Results, so that we can close this issue?

About your timings: 20ms is pretty slow! Perhaps you use inappropriate data-types. Can we discuss this in another thread or via eMail? Could you send me the relevant code via gist?
I can achieve a SweptVolume coll-detection with a map-size of 200x200x100 in 0.56 ms when colliding a BitVoxelList against a BitVoxelMap (on a GTX 960).

Best,
Andreas

wxmerkt, 2017-01-24:
Hi Andreas,
Thanks for the update and apologies for not being in touch. I checked again today and collision times were 0.3ms as well as the return for BitVoxelList against BitVoxelMap was fine - I must have previously measured it including creating the swept volume (which is 2-20ms depending on number of configurations - what are the numbers you've seen for the UR5?). Is there any way to speed-up the creation of swept volumes (my robot's voxelised at 1cm resolution for the 5cm map - when I voxelise it at 5cm I see issues with it loading correctly/not displaying in gpu_voxels_visualiser).

Thanks, best wishes,
Wolfgang

closed 2017-01-24 by wxmerkt

Squelsh, 2017-01-24:
Good to hear that things are fine now!

Well, the creation of the Sweeps strictly depends on the number of interpolation steps and the size of the robot-model-point-clouds.
For the voxelization resolution you should chose a factor < ( Map-Resolution / sqrt(2) ), otherwise you will run into discretization errors when rotating the pointclouds.
So at 5cm grid-size a model resolution of 2.5cm is sufficient.
And the required number of configurations can simply be derived from the thickness of your robot model and the maximum allowed joint motion for which the single voxelized robot postures still overlap.
Otherwise there is not much you can do, to decrease sweep-creation time.

General remarks: 
1) 5cm map resolution is pretty coarse. On your GTX 970 you should be able to handle billions of Voxels with only a slight runtime increase. At your current (very low) voxel-count the context switching delays between CPU and GPU may dominate the actual calculation times...
So why not increase resolution?

2) Why is the speed for the sweep-creation important to you? I always see this as an offline step...

3) If you can tell me, what big picture you are trying to achieve, I can perhaps help you better.

Cheers,
Andreas