fzi-forschungszentrum-informatik/gpu-voxels #46: cmake error at example_how_to_link

maxtom, 2017-06-21:
I follow the setup instruction, and anything is ok, but when I try to test the example_how_to_link, I got the following errors:

-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
-- GVL include dirs are: /home/i/codebase/Topological-SLAM/gpu-voxels/packages/gpu_voxels/src
CMake Error at CMakeLists.txt:47 (target_link_libraries):
  The LINK_PUBLIC or LINK_PRIVATE option must appear as the second argument,
  just after the target name.

-- Configuring incomplete, errors occurred!

Squelsh, 2017-06-21:
Hi Maxtom,

I never experienced this specific error message. Could you give me a bit more details? What OS are you using? Which CMake version? Perhaps I can recreate the error locally.

Cheers,
Andreas

Squelsh, 2017-08-02:
Please give some more info. Otherwise I will close this issue.

Squelsh, 2017-08-13:
Closed due to no further reaction.

closed 2017-08-13 by Squelsh