fzi-forschungszentrum-informatik/gpu-voxels #47: Raycast ?

nangong, 2017-07-17:
Are there any raycast method provided by this project?

Squelsh, 2017-07-17:
Yes, raycasting is used in the Octree and in the Voxelmap, to determine the free space when you insert measurement data.

nangong, 2017-07-17:
I am current using this projects for implementing 3-d localization. Since I want to get the nearest occupancy nodes (of octree) from a robot in a thousand directions , I need an raycast function just like what is provided in [Octomap](https://github.com/OctoMap/octomap/blob/e19c1f33f92495e546175d64eb88fc40f16b62da/octomap/include/octomap/MapCollection.h).  BTW, I did not find such function ,am I missing something?

Squelsh, 2017-07-17:
Good that you mention your actual goals. I would suggest that you consider our EDF (euclidean distance field) implementation in GPU-Voxels. It calculates the distance between EVERY Voxel and its closest obstacle very efficiently (2093 * 10^6 Voxels / sec). So you could then just query the 1000 Voxels that you are interested in.
The distance field actually gives you a pointer to the closest occupied voxel. There is an example in 
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/examples/DistanceKinectDemo.cpp
which queries two distances only.

For the raycasting have a look at 
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/gpu_voxels/octree/GvlNTree.h#L110
and
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/gpu_voxels/voxelmap/ProbVoxelMap.hpp#L53
Both functions insert PointClouds and finally call kernels that implement the raycasting.

Hope that helps? I wish you good progress with your localization approach! Please keep me updated. I would love to integrate it into GPU-Voxels.

Squelsh, 2017-07-17:
This is the basic raycasting within VoxelMaps:
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/gpu_voxels/voxelmap/kernels/VoxelMapOperations.h#L199

But I am sorry, there is no raycasting implemented in the Octree, as the Octree uses the same function, and only afterwards inserts the results into the Octree.
Implementing an Octree-Raycaster with n*log(n) runtime should be easy. Otherwise you could even use the loadbalanced lookup functions.

nangong, 2017-07-18:
My maps for localization are very large (about 1KM), EDF approach is not possible to store such a large 3d matrix in the memory of GPU.  The raycasting example you mentioned may help, the only problem is  the efficiency. By using a STACK struct , it is more efficient to trace ray in an octree than always access from the root node. An implementation is [Efficient-Sparse-Voxel-Octrees](https://code.google.com/archive/p/efficient-sparse-voxel-octrees/). But the project seems not be maintained any more.

nangong, 2017-07-20:
Since I only want to know the first occupancy node from the origin sensor along the ray, the runtime is only log(n). BTW, the raycast sample you mentioned is helpful.

closed 2017-07-20 by nangong

mcelhennyi, 2020-02-17:
> This is the basic raycasting within VoxelMaps: [...]

Hello, I have a similar need. I have a depth map that I would like to ray cast toward to increment and decrement occupancy along the path from source to point cloud point..

Should I use this "RayCaster"? It seems to be implemented in CPU, maybe I am misreading it?

"insertPointCloudWithFreespaceCalculation" seems to only tell me an explicit free or not, whereas I would instead need a decrement/increment of some occupancy value.

Thanks!