fzi-forschungszentrum-informatik/gpu-voxels #49: How to calculate the bounding box of nodes during traversing?

nangong, 2017-07-26:
Hi, I have created a gpu-voxel octree  and wanted to traverse the whole octree from the root.  Are there any api to calcuate a bounding box of the current accessed node(both InnerNode and LeafNode)?

Squelsh, 2017-07-26:
This is exactly what is happening for visualization: A list of all nodes is generated, together with their bounding boxes. This list is then shared with the visualizer process to draw it in OpenGL.

Look here:
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/gpu_voxels/octree/VisNTree.hpp#L104

This generates a list of Cubes, as defined in here:
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/79a64226e7307e058205f43e375fac6f6c7c768d/packages/gpu_voxels/src/gpu_voxels/vis_interface/VisualizerInterface.h#L87

They have a position and a side_lenght.

nangong, 2017-07-27:
I found  OctreeVoxelID is not  stored in each node. How could I get the OctreeVoxelID of an accessed node when traversing?

nangong, 2017-07-27:
@Squelsh  I found OctreeVoxelID is not stored in each node. How could I get the OctreeVoxelID of an accessed node when traversing?

Squelsh, 2017-07-27:
No, we minimized the memory footprint of the nodes. And the ID is more or less the iterator, so there was no need to store it in the nodes.
Depending on what you plan to do, you could resore the VoxelID from the coordinates (only suitable for non-time-critical tasks) or you have to take a look at the function which extracts the Nodes. There you should find the VoxelIDs.

nangong, 2017-07-27:
My plan is mentioned on this issue [issue#47](https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/issues/47) , i.e find the nearest obstacle along a given ray. When I traverse the octree and get the bounding box of a node, I can check which child is intersected and thus traverse into the child. But now I cannot found where to get the voxelID.

closed 2017-07-31 by nangong