fzi-forschungszentrum-informatik/gpu-voxels #52: Buggy assert  in print() function?

nangong, 2017-08-02:
Hi, I found a possible buggy assert in [kernel_print function](https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/79a64226e7307e058205f43e375fac6f6c7c768d/packages/gpu_voxels/src/gpu_voxels/octree/kernels/kernel_Octree.h), 
`
#line 164
 assert(
  (level == 0 && node.hasStatus(ns_LAST_LEVEL)) || (level != 0 && !node.hasStatus(ns_LAST_LEVEL)));
`
should this line be,
`
 assert(
  (level == 1 && node.hasStatus(ns_LAST_LEVEL)) || (level != 1 && !node.hasStatus(ns_LAST_LEVEL)))
`
Since nodes in level 0 are all leaf nodes , and   in [NTree.hpp](https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/9949ee6721e573d03ed44e48b6630ddd208bcc81/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp),
`
        #line 607
        if (level == 1)
        {
          kernel_clearNodes<InnerNode, true> <<<numBlocks, num_threads_per_block>>>(branching_factor * numNodes, (InnerNode*) nodes );
          HANDLE_CUDA_ERROR(cudaDeviceSynchronize());
          kernel_setNodes<InnerNode, LeafNode, branching_factor> <<<numBlocks, num_threads_per_block>>>(
              D_PTR(d_voxels), num_points, level, D_PTR(nodeCount),
              (InnerNode*) nodes, D_PTR(nodeIds), (LeafNode*) childNodes);
}
`
the above code set ns_LAST_LEVEL flag for nodes in level 1.

Squelsh, 2017-08-02:
Thanks @nangong !! 
From a first look you seem to be right. ns_LAST_LEVEL is not the Leaf-Level, but one level above.
It is used to trigger serial execution of Leaf-Node operations to prevent inefficient full parallelization.
So the assert is buggy.
I will check this again and correct it with an upcoming release (which should come soon).

nangong, 2017-08-03:
@Squelsh  
Moreover , I found both level 0 and level 1 are set ns_LAST_LEVEL flag