fzi-forschungszentrum-informatik/gpu-voxels #54: How to use cuda gdb in gpu-voxel?

nangong, 2017-08-24:
Hi , I want to use cudagdb to debug  gpu-voxel, but after I set the flag 'SET(CUDA_NVCC_FLAGS ${CUDA_NVCC_FLAGS} "-Xcompiler=-Wall -g -G")' in src/CMakeLists.txt   compiles the whole project and run the example/Maps program , the program stuck without any output? How did u debug the project?

Squelsh, 2017-09-01:
To be honest: I never used CUDA GDB.... Only regular GDB.
Did you find out, what the problem was, by now?

Just to be sure: Did you wait long enough after the start, for the JIT compilation of CUDA? That can take up to 2 minutes on some systems...

nangong, 2017-09-26:
I did not find out the reason , after "long enough" (about 24 hours !!!!), the program still not execute....Therefore , I have to use log for debugging kernel functions. :(

Squelsh, 2018-02-09:
Is this issue still active?
Did you find the root of the problem?
Did you try the Nsight Debugger? It's a GUI for Cuda GDB.
https://developer.nvidia.com/nsight-eclipse-edition

Please close the issue, if solved.

cjue, 2018-02-09:
I wasn't able to reproduce the freeze reported by Hiro, but to my understanding it's currently not possible to do single-step debugging with cuda-gdb (or Nsight Debugger) because of compilation issues.

To my current knowledge, adding the "-G" compiler option leads to "PTX JIT compilation failed" errors during runtime. To debug that, it would be interesting to change the way we call NVCC in order to force ptxas to run at compile time.

cjue, 2018-08-28:
Should be fixed by Release 1.2.1 feb56b3, just as issue #73.

We have observed unusually long startup times when running programs compiled with the "-G" option sometimes though.

closed 2018-08-28 by cjue