fzi-forschungszentrum-informatik/gpu-voxels #55: Compile issue on Jetson TX2

dybedal, 2017-09-01:
I'm trying to compile gpu-voxels on an NVIDIA Jetson TX2, but I run into the following error:
```
[ 83%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_visualization/CMakeFiles/gpu_voxels_visualization_core_CUDA_TARGET.dir/gpu_voxels_visualization_core_CUDA_TARGET_generated_Visualizer.cu.o
In file included from /usr/include/c++/5/type_traits:35:0,
                 from /usr/local/include/glm/detail/type_int.hpp:8,
                 from /usr/local/include/glm/fwd.hpp:8,
                 from /usr/local/include/glm/glm.hpp:90,
                 from /home/nvidia/gpu-voxels/packages/gpu_voxels/src/gpu_visualization/Visualizer.h:64,
                 from /home/nvidia/gpu-voxels/packages/gpu_voxels/src/gpu_visualization/Visualizer.cu:16:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
...
```
I tried adding -std=c++11 to the nvcc flags in packages/gpu-voxels/CMakeLists.txt (ln 139), but that did not work and introduced other errors: 
```
[ 52%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/gpu_voxels_helpers_CUDA_TARGET_generated_MetaPointCloud.cu.o
/home/nvidia/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/helpers/cuda_handling.hpp(126): error: no operator "<<" matches these operands
            operand types are: icl_core::logging::ThreadStream << std::stringstream

1 error detected in the compilation of "/tmp/tmpxft_00002911_00000000-5_PointCloud.cpp4.ii".
CMake Error at gpu_voxels_helpers_CUDA_TARGET_generated_PointCloud.cu.o.cmake:266 (message):
  Error generating file
  /home/nvidia/gpu-voxels/build/packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir//./gpu_voxels_helpers_CUDA_TARGET_generated_PointCloud.cu.o
...
```
Do you have any suggestions?

Squelsh, 2017-09-01:
Hi Dybedal,

we managed to compile GPU-Voxels on the TX1 once but did not follow that path any longer.
Our code definitely does not require any C++11 features. My guess would be, that you are using a environment wich is too new (like a new Boost version or something) and that some of the dependencies introduce the C++11 requirements.

But as your error log says something about GLM: Did you patch your GLM version? Perhaps you ran into [this](https://github.com/g-truc/glm/issues/530) problem?

Compiling GPU Voxels on a recent linux is on our agenda currently. So if you fix this problem, please share your findings with us.

Good luck,
Andreas

dybedal, 2017-09-05:
Hi and thanks for your quick reply.

I had already patched the GLM file, so that was not an issue. I managed to compile gpu-voxels without the visualization package, which is a bummer, but at least I can use your library in my project now. 

But in my project I am also using other libraries which require C++11 support (i.e. Stereolabs ZED SDK). When I include <gpu_voxels/GpuVoxels.h> I cannot compile due to the missing implementation of the << operator for `icl_core::logging::ThreadStream << std::stringstream`, as mentioned in the previous post. 

However if I just add this declaration to icl_core_logging/ThreadStream.h:
`ICL_CORE_LOGGING_IMPORT_EXPORT ThreadStream& operator << (ThreadStream& stream, const **std::stringstream&** text);`. 
everything compiles OK, so that seems to be the only issue. Of course the function also needs to be implemented...

I hope adding this will also make it possible to compile everything (including visualization) with C++11 support from the start, but I haven't tested that yet.

BR,
Joacim

cjue, 2017-09-05:
Hi Joacim,

sorry for this: we already fixed this bug internally. 
I'll release a bugfix release until the end of the week.

### Patch

> diff --git a/src/gpu_voxels/helpers/cuda_handling.hpp b/src/gpu_voxels/helpers/cuda_handling.hpp
> index ae7f3b7..5f171d5 100644
> --- a/src/gpu_voxels/helpers/cuda_handling.hpp
> +++ b/src/gpu_voxels/helpers/cuda_handling.hpp
> @@ -123,7 +123,7 @@ cudaError_t cuPrintDeviceArray(T* dev_array, unsigned int* device_array_size)
>    {
>      std::stringstream s;
>      s << i << ": " << host_array[i];
> -    LOGGING_INFO(Gpu_voxels_helpers, s << endl);
> +    LOGGING_INFO(Gpu_voxels_helpers, s.str() << endl);
>    }
>    LOGGING_INFO(Gpu_voxels_helpers, endl);
>    delete[] host_array;

dybedal, 2017-09-05:
Thanks, that worked like a charm!

Not sure how useful this is to you, but this is what I had to do to be able to compile the visualization package with C++11 support:

## PATCH:
File: gpu_voxels/packages/gpu_voxels/src/gpu_voxels/helpers/cuda_handling.hpp ln 126:
`	LOGGING_INFO(Gpu_voxels_helpers, s.str() << endl);`

## ENABLE C++11 SUPPORT:
File: gpu-voxels/CMakeLists.txt (top):
```
CMAKE_MINIMUM_REQUIRED(VERSION 3.1)
SET(CMAKE_CXX_STANDARD 11)
```
## Set CUDA computability
File: gpu-voxels/packages/gpu_voxels/CMakeLists.txt (ln 134):
`SET(ICMAKER_CUDA_COMPUTE_VERSION 60) #For Jetson TX2`

## Workaround for GLM error "...this file requires C+11..." (using latest GLM from github):
File: gpu-voxels/packages/gpu_voxels/CMakeLists.txt (ln 139):
`SET(CUDA_NVCC_FLAGS ${CUDA_NVCC_FLAGS} "-Xcompiler=-Wall -std=c++11")`

## isnan must be ::isnan:
[Reference](https://stackoverflow.com/questions/33770374/why-is-isnan-ambiguous-and-how-to-avoid-it)
File: gpu_voxels/src/gpu_voxels/octree/PointCloud.cu (ln 411):
`    return !::isnan(v.x) & !::isnan(v.y) & !::isnan(v.z);`

## static const float --> static constexpr float:
[Reference](https://stackoverflow.com/questions/9141950/initializing-const-member-within-class-declaration-in-c)
File: gpu_voxels/src/gpu_voxels/octree/load_balancer/AbstractLoadBalancer.h (ln 59):
`  static constexpr float DEFAULT_IDLE_THESHOLD = 2.0f/3.0f;`

## #error "GLM: GLM_GTX_transform is an experimental extension and may change in the future. Use `#define GLM_ENABLE_EXPERIMENTAL` before including it, if you really want to use it."
File: gpu-voxels/packages/gpu_voxels/src/gpu_visualization/Primitive.h (ln 32):
`#define GLM_ENABLE_EXPERIMENTAL`

FYI: I do not have ROS installed at the moment, so I have not tried to compile with ROS support. The following configuration was built:
```
-- ------------------------ GPU Voxels configuration ------------------------
--   
-- [OK]      Building GPU-Voxels. Found CUDA.
-- [OK]      Building GPU-Voxels with visualization. GLEW, GLM, OpenGL and GLUT were found.
-- [WARNING] Building GPU-Voxels without ROS connections. ROS not found.
-- [WARNING] Building GPU-Voxels without URDF support. Could not find the following packages:
--     urdfdom
--     orocos_kdl
--     kdl_parser
--     ROS
-- [OK]      Building GPU-Voxels with Kinect support. OpenNI was found.
-- [OK]      Building GPU-Voxels with PCL interfaces. PCL found.
--   
-- -------------------- END of GPU Voxels configuration ---------------------
```

dybedal, 2017-09-11:
Update: Installed ROS Kinetic and was able to build with ROS and URDF support, but had to apply your fix in another file as well:

```
diff --git a/packages/gpu_voxels/src/gpu_voxels/robot/urdf_robot/robot_link.cpp b/packages/gpu_voxels/src/gpu_voxels/robot/urdf_robot/robot_link.cpp
index f4f7221..1f75a8a 100644
--- a/packages/gpu_voxels/src/gpu_voxels/robot/urdf_robot/robot_link.cpp
+++ b/packages/gpu_voxels/src/gpu_voxels/robot/urdf_robot/robot_link.cpp
@@ -155,7 +155,7 @@ RobotLink::RobotLink(Robot* robot,
     desc << "  This link has NO geometry.";
   }
 
-  LOGGING_DEBUG_C(RobotLog, RobotLink, desc << endl);
+  LOGGING_DEBUG_C(RobotLog, RobotLink, desc.str() << endl);
 
 }

```

Squelsh, 2017-09-13:
Hi dybedal,
thanks a lot for sharing those findings! C++11 is really relevant.
Why don't you create a clone of GPU-Voxels and push your fixes?
So we can either create a C++11 branch or just merge them into the master.

cjue, 2018-02-05:
All the mentioned fixes should be included in the new 1.1 Release: https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/commit/d9914ac10e82e50415af1cf7f175a406f99b4cd4

The one exception is "GLM_ENABLE_EXPERIMENTAL". I did not yet encounter the corresponding GLM is myself. Which GLM version is this related to?

Changliu52, 2018-02-07:
I am able to compile it in tx2 with the latest commit in [https://github.com/g-truc/glm.git](url), be53cebcd3ec3366e30f520bfaa35991274fc390 with GLM_ENABLE_EXPERIMENTAL set, with ros and urdf

cjue, 2018-02-07:
Thanks for the feedback! The GLM_ENABLE_EXPERIMENTAL define will be added to Primitive.h for release 1.1.2.

Strange though, that it is not treated as a compile error on Ubuntu 14.04 and 16.04 GLM.

closed 2018-02-07 by cjue