fzi-forschungszentrum-informatik/gpu-voxels #59: No examples built after make, no ros or urdf, ubuntu 16.04

AndreasMegatranus, 2017-11-14:
I've done the installation (on Ubuntu 16.04), but after the make, the build/bin has executables for testing icl_core, but not for the examples.

cmake indicates:

-- --------------------------------------------------------------------------
-- ------------------------ GPU Voxels configuration ------------------------
--   
-- [OK]      Building GPU-Voxels. Found CUDA.
-- [OK]      Building GPU-Voxels with visualization. GLEW, GLM, OpenGL and GLUT were found.
-- [WARNING] Building GPU-Voxels without ROS connections. ROS not found.
-- [WARNING] Building GPU-Voxels without URDF support. Could not find the following packages:
--     urdfdom
--     orocos_kdl
--     kdl_parser
--     ROS
-- [WARNING] Building GPU-Voxels without Kinect support. OpenNI not found.
-- [OK]      Building GPU-Voxels with PCL interfaces. PCL found.
--   
-- -------------------- END of GPU Voxels configuration ---------------------
-- --------------------------------------------------------------------------

Here is CMakeError.log:

Determining if the pthread_create exist failed with the following output:
Change Dir: /home/andreas/GPU-Voxels/gpu-voxels/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_edd0f/fast"
/usr/bin/make -f CMakeFiles/cmTC_edd0f.dir/build.make CMakeFiles/cmTC_edd0f.dir/build
make[1]: Entering directory '/home/andreas/GPU-Voxels/gpu-voxels/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_edd0f.dir/CheckSymbolExists.c.o
/usr/bin/cc    -Wall -Wno-long-long -msse -msse2 -ffunction-sections     -o CMakeFiles/cmTC_edd0f.dir/CheckSymbolExists.c.o   -c /home/andreas/GPU-Voxels/gpu-voxels/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_edd0f
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_edd0f.dir/link.txt --verbose=1
/usr/bin/cc    -Wall -Wno-long-long -msse -msse2 -ffunction-sections      CMakeFiles/cmTC_edd0f.dir/CheckSymbolExists.c.o  -o cmTC_edd0f -rdynamic 
CMakeFiles/cmTC_edd0f.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text.main+0x16): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_edd0f.dir/build.make:97: recipe for target 'cmTC_edd0f' failed
make[1]: *** [cmTC_edd0f] Error 1
make[1]: Leaving directory '/home/andreas/GPU-Voxels/gpu-voxels/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_edd0f/fast' failed
make: *** [cmTC_edd0f/fast] Error 2

File /home/andreas/GPU-Voxels/gpu-voxels/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/andreas/GPU-Voxels/gpu-voxels/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_660bc/fast"
/usr/bin/make -f CMakeFiles/cmTC_660bc.dir/build.make CMakeFiles/cmTC_660bc.dir/build
make[1]: Entering directory '/home/andreas/GPU-Voxels/gpu-voxels/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_660bc.dir/CheckFunctionExists.c.o
/usr/bin/cc    -Wall -Wno-long-long -msse -msse2 -ffunction-sections  -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_660bc.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.5/Modules/CheckFunctionExists.c
Linking C executable cmTC_660bc
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_660bc.dir/link.txt --verbose=1
/usr/bin/cc    -Wall -Wno-long-long -msse -msse2 -ffunction-sections  -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_660bc.dir/CheckFunctionExists.c.o  -o cmTC_660bc -rdynamic -lpthreads 
/usr/bin/ld: cannot find -lpthreads
[... 80 lines cut ...]
/usr/bin/c++    -I/usr/include/qt4  -Wall -Wno-long-long -msse -msse2 -ffunction-sections     -o CMakeFiles/cmTC_74bbc.dir/CheckSymbolExists.cxx.o -c /home/andreas/GPU-Voxels/gpu-voxels/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx
/home/andreas/GPU-Voxels/gpu-voxels/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx: In function ‘int main(int, char**)’:
/home/andreas/GPU-Voxels/gpu-voxels/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx:8:19: error: ‘Q_WS_MAC’ was not declared in this scope
   return ((int*)(&Q_WS_MAC))[argc];
                   ^
CMakeFiles/cmTC_74bbc.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_74bbc.dir/CheckSymbolExists.cxx.o' failed
make[1]: *** [CMakeFiles/cmTC_74bbc.dir/CheckSymbolExists.cxx.o] Error 1
make[1]: Leaving directory '/home/andreas/GPU-Voxels/gpu-voxels/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_74bbc/fast' failed
make: *** [cmTC_74bbc/fast] Error 2

File /home/andreas/GPU-Voxels/gpu-voxels/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx:
/* */
#include <QtCore/qglobal.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef Q_WS_MAC
  return ((int*)(&Q_WS_MAC))[argc];
#else
  (void)argc;
  return 0;
#endif
}

cjue, 2017-11-16:
Hi, can you please run CMake in a fresh build directory and provide the full output?

Unfortunately the error log from above does not have enough context to find a solution.
The release version is tested and should work&#x2122; on vanilla Ubuntu 16.04.

AndreasMegatranus, 2017-11-16:
OK, I did that.  

Terminal output from the cmake is attached below in the file AGH_CMake_Terminal_Output_11162017.txt.  I've also attached the file CMakeOutput.Log, which is different.

I noticed that right at the beginning (of the first file), there is a warning about FindThreads.cmake being shadowed.  After that, there are some checks where it doesn't find pthread_create on the first few tries, but then finds it at the end.

[AGH_CMake_Terminal_Output_11162017.txt](https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/files/1478969/AGH_CMake_Terminal_Output_11162017.txt)

[CMakeOutput.log](https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/files/1478984/CMakeOutput.log)

AndreasMegatranus, 2017-11-16:
Sorry, I accidentally closed the issue in my last comment.

cjue, 2017-11-16:
Hi Andreas,

the main problem is the missing TinyXML. The package "libtinyxml-dev" should fix these problems:

```
-- Could NOT find TinyXML (missing:  TINYXML_INCLUDE_DIR TINYXML_LIBRARIES) 
...
-- Warning: icl_core_config -- not building library, missing [tinyxml].
-- Warning: icl_core_logging -- not building library, missing [icl_core_config].
...
-- Warning: gpu_voxels_core -- not building library, missing [icl_core_config;icl_core_logging;gpu_voxels_logging;gpu_voxels_helpers;gpu_voxels_primitive_array].

```

Let me know if this fixes the problem :)

cjue, 2017-11-17:
The added dependency is now mentioned on our website: 
http://gpu-voxels.org/documentation/prerequisites/

AndreasMegatranus, 2017-11-20:
I installed libtinyxml-dev.  It seems to be trying harder now, but failed during the make with the following message:

[ 72%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/octree/CMakeFiles/gpu_voxels_octree_CUDA_TARGET.dir/gpu_voxels_octree_CUDA_TARGET_generated_Octree.cu.o
In file included from /home/andreas/GPU-Voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/Octree.cu:15:0:
/home/andreas/GPU-Voxels/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp:56:43: fatal error: thrust/system/cuda/detail/cub.h: No such file or directory
compilation terminated.
CMake Error at gpu_voxels_octree_CUDA_TARGET_generated_Octree.cu.o.cmake:207 (message):
  Error generating
  /home/andreas/GPU-Voxels/gpu-voxels/build/packages/gpu_voxels/src/gpu_voxels/octree/CMakeFiles/gpu_voxels_octree_CUDA_TARGET.dir//./gpu_voxels_octree_CUDA_TARGET_generated_Octree.cu.o

Squelsh, 2017-11-23:
Did you check for the missing file (thrust/system/cuda/detail/cub.h)?
Which version of CUDA are you using? Perhaps the include directories for CUB have changed...

We will investigate.

cjue, 2017-11-23:
Hi, 

the error message lists Cuda 9.0 as the version used.
I am currently testing this myself with Cuda 9.

cjue, 2017-11-23:
Hi,
the problem stems from the changes to Thrust in Cuda 9. 
More specifically: Cuda 9's Thrust includes CUB in a different way. 

I'm working on a simple patch for this.

cjue, 2017-11-23:
I found a solution for the Thrust cub namespace changes. Now another problem popped up, where a Cuda9 change breaks Eigen3:

usr/local/cuda/include/crt/common_functions.h:64:24: error: token ""__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead.""
https://devtalk.nvidia.com/default/topic/1026622/nvcc-can-t-compile-code-that-uses-eigen/

I'll build a workaround that does not include patching Eigen3...

cjue, 2017-11-23:
Found a fix for the Eigen3 - Cuda9 incompatibility.
I will need to run some further testing to make sure the combined fixes don't break a lot of stuff under Cuda9 or Cuda8.

The simplest workaround is to use CUDA 8. 

To force CMake to use that version you can use the CUDA_PATH environment variable:
` export CUDA_PATH=/usr/local/cuda-8.0`

AndreasMegatranus, 2017-11-23:
OK, I will try with CUDA 8

AndreasMegatranus, 2017-11-28:
So, I purged CUDA 9 from my computer, installed CUDA 8, uninstalled all the GPU-Voxel dependencies and then reinstalled them.  The cmake succeeded, but the make got 88% finished and then failed trying to build gpu_voxels_visualization_core.  The error output from the make is as follows:

[ 87%] Built target test_gpu_voxels_core
[ 88%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_visualization/CMakeFiles/gpu_voxels_visualization_core_CUDA_TARGET.dir/kernels/gpu_voxels_visualization_core_CUDA_TARGET_generated_VoxelMapVisualizerOperations.cu.o
[ 88%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_visualization/CMakeFiles/gpu_voxels_visualization_core_CUDA_TARGET.dir/gpu_voxels_visualization_core_CUDA_TARGET_generated_Visualizer.cu.o
/usr/local/cuda-8.0/include/thrust/detail/allocator/allocator_traits.inl(110): warning: calling a __host__ function from a __host__ __device__ function is not allowed
          detected during:
            instantiation of "thrust::detail::enable_if<thrust::detail::allocator_traits_detail::has_member_construct2<Alloc, T, Arg1>::value, void>::type thrust::detail::allocator_traits_detail::construct(Alloc &, T *, const Arg1 &) [with Alloc=std::allocator<gpu_voxels::visualization::colorPair>, T=gpu_voxels::visualization::colorPair, Arg1=gpu_voxels::visualization::colorPair]" 
(278): here
            instantiation of "void thrust::detail::allocator_traits<Alloc>::construct(thrust::detail::allocator_traits<Alloc>::allocator_type &, T *, const Arg1 &) [with Alloc=std::allocator<gpu_voxels::visualization::colorPair>, T=gpu_voxels::visualization::colorPair, Arg1=gpu_voxels::visualization::colorPair]" 
/usr/local/cuda-8.0/include/thrust/detail/allocator/copy_construct_range.inl(54): here
            instantiation of "void thrust::detail::allocator_traits_detail::copy_construct_with_allocator<Allocator, InputType, OutputType>::operator()(Tuple) [with Allocator=std::allocator<gpu_voxels::visualization::colorPair>, InputType=gpu_voxels::visualization::colorPair, OutputType=gpu_voxels::visualization::colorPair, Tuple=thrust::detail::tuple_of_iterator_references<gpu_voxels::visualization::colorPair &, gpu_voxels::visualization::colorPair &, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type>]" 
/usr/local/cuda-8.0/include/thrust/detail/function.h(60): here
            instantiation of "Result thrust::detail::wrapped_function<Function, Result>::operator()(const Argument &) const [with Function=thrust::detail::allocator_traits_detail::copy_construct_with_allocator<std::allocator<gpu_voxels::visualization::colorPair>, gpu_voxels::visualization::colorPair, gpu_voxels::visualization::colorPair>, Result=void, Argument=thrust::detail::tuple_of_iterator_references<gpu_voxels::visualization::colorPair &, gpu_voxels::visualization::colorPair &, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type>]" 
/usr/local/cuda-8.0/include/thrust/system/detail/sequential/for_each.h(56): here
            instantiation of "InputIterator thrust::system::detail::sequential::for_each(thrust::system::detail::sequential::execution_policy<DerivedPolicy> &, InputIterator, InputIterator, UnaryFunction) [with DerivedPolicy=thrust::system::cpp::detail::tag, InputIterator=thrust::zip_iterator<thrust::tuple<thrust::detail::normal_iterator<gpu_voxels::visualization::colorPair *>, gpu_voxels::visualization::colorPair *, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type>>, UnaryFunction=thrust::detail::allocator_traits_detail::copy_construct_with_allocator<std::allocator<gpu_voxels::visualization::colorPair>, gpu_voxels::visualization::colorPair, gpu_voxels::visualization::colorPair>]" 
/usr/local/cuda-8.0/include/thrust/detail/for_each.inl(44): here
            [ 4 instantiation contexts not shown ]
            instantiation of "thrust::detail::contiguous_storage<T, Alloc>::iterator thrust::detail::contiguous_storage<T, Alloc>::uninitialized_copy(InputIterator, InputIterator, thrust::detail::contiguous_storage<T, Alloc>::iterator) [with T=gpu_voxels::visualization::colorPair, Alloc=std::allocator<gpu_voxels::visualization::colorPair>, InputIterator=thrust::detail::normal_iterator<gpu_voxels::visualization::colorPair *>]" 
/usr/local/cuda-8.0/include/thrust/detail/vector_base.inl(833): here
            instantiation of "void thrust::detail::vector_base<T, Alloc>::fill_insert(thrust::detail::vector_base<T, Alloc>::iterator, thrust::detail::vector_base<T, Alloc>::size_type, const T &) [with T=gpu_voxels::visualization::colorPair, Alloc=std::allocator<gpu_voxels::visualization::colorPair>]" 
/usr/local/cuda-8.0/include/thrust/detail/vector_base.inl(593): here
            instantiation of "void thrust::detail::vector_base<T, Alloc>::insert(thrust::detail::vector_base<T, Alloc>::iterator, thrust::detail::vector_base<T, Alloc>::size_type, const T &) [with T=gpu_voxels::visualization::colorPair, Alloc=std::allocator<gpu_voxels::visualization::colorPair>]" 
/usr/local/cuda-8.0/include/thrust/detail/vector_base.inl(581): here
[... cut ...]
make[2]: *** [packages/gpu_voxels/src/gpu_visualization/CMakeFiles/gpu_voxels_visualization_core_CUDA_TARGET.dir/gpu_voxels_visualization_core_CUDA_TARGET_generated_Visualizer.cu.o] Error 1
CMakeFiles/Makefile2:2100: recipe for target 'packages/gpu_voxels/src/gpu_visualization/CMakeFiles/gpu_voxels_visualization_core.dir/all' failed
make[1]: *** [packages/gpu_voxels/src/gpu_visualization/CMakeFiles/gpu_voxels_visualization_core.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

AndreasMegatranus, 2017-11-28:
I looked in the build/bin directory.  This time, it did build test_gpu_voxels_core, and this seems to run as expected.

kne1p, 2017-11-29:
Hi Andreas,

looks like you found another known issue:
The GLM library in 16.04 hast a bug that is responsible for this compile error.

I'll have more details in how to fix this later.

Also I'll update the website and readme to warn about this issue. Sorry for not documenting this earlier.

kne1p, 2017-11-29:
So, the GLM issue was this one: https://github.com/g-truc/glm/issues/530

We mentioned it briefly in the 0.9 release changelog and in issue #55, but this should bei more visible.

The fix ist to patch GLM as mentioned in https://github.com/g-truc/glm/issues/530

The problem ist caused by a NVCC problem when compiling 16.04's GLM.

AndreasMegatranus, 2017-11-29:
OK, thanks so much.  I will try the patch.

AndreasMegatranus, 2017-11-30:
The patch worked!  Thank you so much for all your help.  I now see appropriate example executables in the build/bin directory.

cjue, 2018-01-13:
Works with Cuda 8. 

The remaining problems with Cuda 9 are handled in issue #63

closed 2018-01-13 by cjue