fzi-forschungszentrum-informatik/gpu-voxels #60: run robot_vs_environment, test_gpu_voxels_core, meet problems

tongbaiming, 2017-12-06:
Hi, when I run the above tow examples, it reports problems like this:
/home/tbm/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/EnvironmentNodes.h:289: void gpu_voxels::NTree::Environment::InnerNode::setChildPtr(void *): block: [22,0,0], thread: [31,0,0] Assertion `uint64_t(child) < (uint64_t(1) << uint64_t(8 * (sizeof(m_child_high) + sizeof(m_child_middle) + sizeof(m_child_low))))` failed.
<2017-12-07 00:04:38.246> CudaLog(Error)::cuHandleError: device-side assert triggered in /home/tbm/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp on line 623.
<2017-12-07 00:04:38.246> CudaLog(Error)::cuHandleError: device-side assert triggered in /home/tbm/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp on line 540.
terminate called after throwing an instance of 'thrust::system::detail::bad_alloc'
  what():  std::bad_alloc: device-side assert triggered

what is the problems?

cjue, 2017-12-12:
Hi, I'll look into it in the coming days.

Just two questions: what is the Cuda Version and the GPU model you are using?

cjue, 2018-01-13:
My guess is that there was insufficient video memory. 

Issue closed until there is more information.

zhouq0725, 2018-01-29:
hi, cjue.
I have 3 configurations, both use Ubuntu 16.04 and nvidia driver 387.34 to run:

distance_kinect_demo: 
GTX 1050(2GB) , CUDA 9.1.85-1, **"segment fault (core dumped)"**
GTX 1080Ti (11GB), CUDA 8.0, **OK**
collisions:
GTX 1050(2GB) , CUDA 9.1.85-1, **Assertion uint64_t(child) < (uint64_t(1) << uint64_t(8 * (sizeof(m_child_high) + sizeof(m_child_middle) + sizeof(m_child_low)))) failed.**
GTX 1080Ti (11GB), CUDA 8.0, **Assertion uint64_t(child) < (uint64_t(1) << uint64_t(8 * (sizeof(m_child_high) + sizeof(m_child_middle) + sizeof(m_child_low)))) failed.**

cjue, 2018-01-29:
Hi both of you, sorry for prematurely closing this issue.

There are known issue with Cuda 9.1 and Pascal GPUs regarding voxellists and octrees.
We are currently working on resolving those.

The distance_kinect_demo is probably unrelated and might be connected to exceeding the 2GB VRAM on that device. I would be grateful if you could re-run that demo with the release I am going to publish this week.

giorgionicola, 2018-02-20:
Hi @cjue  
I'm having exactly the same issue with a GTX1050 and CUDA 8.0 .
I supposed that the problem it has not been fixed or that the problem is the GPU

cjue, 2018-02-20:
Hi @giorgionicola,

the problem is definitely with some interaction of GPU-Voxels Octree code and the Pascal GPUs (GTX 1050, 1070, 1080, TitanXp).

We are still debugging the cause of these problems. Let us know if there are any other problems regarding the non-octree parts of GPU-Voxels!

cjue, 2018-02-20:
@zhouq0725: Could you please try rerunning the distance_kinect_demo with the current GPU-Voxels release?

If that problem was related to CUDA 9.1, it should be fixed now.

AlejoDiaz49, 2018-08-06:
Hi all,

My configuration is a GTX 1050 with 4GB and CUDA 9.1.85. I am having the same issue with some of the examples.

 - heightmp_loader
 - octree_provider
 - robot_vs_enviroment

`/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/EnvironmentNodes.h:289: void gpu_voxels::NTree::Environment::InnerNode::setChildPtr(void *): block: [3,0,0], thread: [233,0,0] Assertion uint64_t(child) < (uint64_t(1) << uint64_t(8 * (sizeof(m_child_high) + sizeof(m_child_middle) + sizeof(m_child_low)))) failed.`

The **test_gpu_voxels_core** test returns 5 errors, and it shows the same one as in the others examples. I didn't launch the examples that require the kinect or ROS.
 
Did you found any solution?
Thanks in advanced.

jlllweco, 2018-08-17:
Hello everyone,
I'm having the same error.Here is my setup:
 uname -a
`Linux shupeng-Precision-Tower-5810 4.15.0-32-generic #35~16.04.1-Ubuntu SMP Fri Aug 10 21:54:34 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`
nvcc --version
`nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2017 NVIDIA Corporation
Built on Fri_Sep__1_21:08:03_CDT_2017
Cuda compilation tools, release 9.0, V9.0.176`
nvidia-smi
`Fri Aug 17 11:17:34 2018       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 384.130                Driver Version: 384.130                   |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 1080    Off  | 00000000:03:00.0  On |                  N/A |
|  0%   43C    P8     9W / 198W |    694MiB /  8105MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0       921      G   /usr/lib/xorg/Xorg                           196MiB |
|    0      1577      G   /opt/teamviewer/tv_bin/TeamViewer             19MiB |
|    0      1804      G   compiz                                       164MiB |
|    0      2019      G   fcitx-qimpanel                                 8MiB |
|    0     19540      G   ...-token=8D86257EA459995611191BCA638AEC4C   106MiB |
|    0     20198      C   ./raycasting_sensor_example                  193MiB |
+-----------------------------------------------------------------------------+`

and the error went like this:
`./test_gpu_voxels_core`
`===========================================================
=      Please run this Benchmark at least two times       =
=   to avoid measuring just in time compilation effects   =
===========================================================
Running 49 test cases...
WARNING: Logging Instance is null, did you initialize the logging framework?
You should initialize the logging framework at the beginning of your program. This will also enable setting the log level on the command line.
DEBUG distance test nrvox=10000, dimX=128, dimY=128, dimZ=131
... JFA distance done
<2018-08-17 10:30:25.379> Performance(Info)::print: ::jumpFlood3D done: 8.685 ms
...PBA distance done
<2018-08-17 10:30:25.381> Performance(Info)::print: ::parallelBanding3D done: 2.136 ms
<2018-08-17 10:30:25.381> VoxelmapLog(Info)::exactDistances3D: #obstacles: 10000
<2018-08-17 10:30:25.381> VoxelmapLog(Info)::exactDistances3D: grid: 501x131, cMAX_NR_OF_BLOCKS: 65535 threads: 1024
...naive exact distance done
<2018-08-17 10:30:25.649> Performance(Info)::print: ::exactDistances3D done: 267.962 ms
<2018-08-17 10:30:25.649> Performance(Info)::print: ::exactDistances3D done: 268.031 ms
<2018-08-17 10:30:25.657> Performance(Info)::print: ::differences3D done: 8.344 ms
<2018-08-17 10:30:25.729> Performance(Info)::print: ::differences3D done: 3.087 ms
<2018-08-17 10:30:25.802> Performance(Info)::print: ::differences3D done: 3.08 ms
DEBUG distance_extraction test dimX=64, dimY=64, dimZ=64
d_voxels[0] = (x, y, z) = (0, 0, 0)

d_voxels[1] = (x, y, z) = (10, 0, 0)

d_voxels[2] = (x, y, z) = (36, 1, 0)

d_voxels[3] = (x, y, z) = (40, 15, 0)

d_voxels[4] = (x, y, z) = (16, 28, 2)

insertTest()
create octree....`

`/home/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/octree/EnvironmentNodes.h:289: void gpu_voxels::NTree::Environment::InnerNode::setChildPtr(void *): block: [54,0,0], thread: [0,0,0] Assertion `uint64_t(child) < (uint64_t(1) << uint64_t(8 * (sizeof(m_child_high) + sizeof(m_child_middle) + sizeof(m_child_low))))` failed.
/home/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/octree/EnvironmentNodes.h:289: void gpu_voxels::NTree::Environment::InnerNode::setChildPtr(void *): block: [54,0,0], thread: [1,0,0] Assertion `uint64_t(child) < (uint64_t(1) << uint64_t(8 * (sizeof(m_child_high) + sizeof(m_child_middle) + sizeof(m_child_low))))` failed.
/home/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/octree/EnvironmentNodes.h:289: void gpu_voxels::NTree::Environment::InnerNode::setChildPtr(void *): block: [54,0,0], thread: [2,0,0] Assertion `uint64_t(child) < (uint64_t(1) << uint64_t(8 * (sizeof(m_child_high) + sizeof(m_child_middle) + sizeof(m_child_low))))` failed.`

Is this package support CUDA9.0? The compiling process is fine and executable files can be achieved. But there must be something wrong with CUDA.

When I try to use it as a library in ZED MINI SDK which only supports CUDA 9.0, the error went like this:
`[ 96%] Built target zed_wrapper_generate_messages
/usr/bin/ld: CMakeFiles/DistanceKinectDemo.dir/src/DistanceKinectDemo.cpp.o: undefined reference to symbol 'cudaMemcpy@@libcudart.so.9.0'
/usr/local/cuda-9.0/lib64/libcudart.so.9.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
[... cut ...] 5: recipe for target '/home/shupeng/zed-ws/devel/lib/zed_wrapper/DistanceKinectDemo' failed
make[2]: *** [/home/shupeng/zed-ws/devel/lib/zed_wrapper/DistanceKinectDemo] Error 1
CMakeFiles/Makefile2:2535: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/DistanceKinectDemo.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/DistanceKinectDemo.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed`
Thanks a lot.

Nyaruko, 2018-08-18:
Ubuntu 16.04, cuda 9.0, I have the exact problem as stated by jlllweco.

cjue, 2018-08-28:
The underlying error leading to the failed assertion is fixed by Release 1.2.1 feb56b3.

Please let me know if you have any further issues.

closed 2018-08-28 by cjue

jlllweco, 2018-08-29:
Thanks for your help. I am using gpuvoxel and pcl on TX2.But have the problem like this. https://stackoverflow.com/questions/52057866/undefined-reference-to-symbol-zn3pcl7console5printens0-15verbosity-levelepkcz?from=singlemessage Would you please helping me?Thank you so much. YX L 邮箱：646281289@qq.com 签名由 网易邮箱大师 定制 On 08/29/2018 01:41, Christian Jülg wrote: The underlying error leading to the failed assertion is fixed by Release 1.2.1 feb56b3. Please let me know if you have any further issues.

cjue, 2018-08-29:
@jlllweco: do you get this error only when compiling your own code?

Did adding PCL like this to CMakeLists.txt help? https://stackoverflow.com/a/52082292

jlllweco, 2018-08-29:
yes, of course I have.But there is something conflict between PCL and gpu voxel .It is ok when I use them in two different ROS node.But it doesn't work when I use them together.Can i send you my progect or CMakelists? The aim of me is to get EDT mapping to use GPU . It is really nice if you can help me.Have any try to use GPUVOXEL in ROS? YX L 邮箱：646281289@qq.com 签名由 网易邮箱大师 定制 On 08/30/2018 00:47, Christian Jülg wrote: @jlllweco: do you get this error only when compiling your own code? Did adding PCL like this to CMakeLists.txt help? https://stackoverflow.com/a/52082292

jlllweco, 2018-08-30:
Sorry ,what I use is the example here.The only thing I want to get is the EDT MAPPING by GPU. It get pointcloud from camera and got EDT MAPPING.It doesn't work on TX2. YX L 邮箱：646281289@qq.com 签名由 网易邮箱大师 定制 On 08/30/2018 01:23, Christian Jülg wrote: Hi, as mentioned here, PCL 1.7 is incompatible with C++11: https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/gvl_ompl_planning/README.txt If you are compiling GPU-Voxels with C++11 support and currently use PCL1.7, please try using pcl1.8, either through a package or by building it yourself. Are you using PCL1.7 and C++11?

jlllweco, 2018-08-30:
Thank you so much if you can help me! I s there any one use EDT mapping on TX2 or ROS before? 

https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/gvl_ompl_planning/README.txt

cjue, 2018-08-30:
Yes, there have been people using all of this on Jetson Boards and in combination with ROS.

I'll be on vacation for the rest of the week and therefore won't be able to respond during that time.

ilia-sheremet, 2018-09-25:
Solved on mine after the update