fzi-forschungszentrum-informatik/gpu-voxels #64: Consistency in PointCloud interfaces

Squelsh, 2018-01-14:
We should improve the API of all point cloud loaders and constructors and offer the same set of parameters.
Some API calls allow you to shift and scale the cloud while loading, some other calls don't.
Also there should not only be the option to translate but also to rotate a cloud while loading it from a file (the API was implemented before we had the matrix calculations in the math helpers).

Example: 
The high level API offers: 
```
bool GpuVoxels::insertPointCloudFromFile(const std::string map_name, const std::string path, const bool use_model_path,
                                const BitVoxelMeaning voxel_meaning, const bool shift_to_zero = false,
                                const Vector3f &offset_XYZ = Vector3f(), const float scaling = 1.0);
```
while the CTOR of PointCloud only has no options at all:

`PointCloud::PointCloud(const std::string &path_to_file, bool use_model_path = true);`

But this is a low prio improvement.

atleaa, 2018-04-27:
A somewhat related question:
Is there a way to scale the binvox files inserted as part of a urdf robot?
If not I will need to find a way to scale the actual binvox or stl files.

Squelsh, 2018-04-27:
You can scale the binvox files during the rasterization process. Therefore I created the voxelize.sh script, which has a scaling parameter.
Translation and rotation defined in the .urdf files is already supported. Scaling unfortunately not.

atleaa, 2018-04-27:
Thanks for the feedback. No wonder I couldn’t find it then. I’ll try the scaling of the binvox files.