fzi-forschungszentrum-informatik/gpu-voxels #67: No robot self-collision check

bsaund, 2018-03-14:
Currently I have not found a nice way to do self-collisions using gpu_voxels. This seems challenging because
1) Adjacent links may occupy the same voxels due to discretization, yet this should not count as a collision
2) It is inefficient and cumbersome to store separate maps for each link and selectively collide the relevant links that may cause self collisions.

Any suggestions?

Squelsh, 2018-03-14:
Hi Brad,
unfortunately you are right: Currently the situation is all but optimal. We had that feature in one of the very early versions of GPU-Voxels but it got lost again.

What you would have to do is to write an insertion kernel which uses Bitvector Voxels to represent the robot, and then identify each part of your robot through an individual ID. And while inserting, you have to check for already occupied voxels. Through the IDs you could mask and ignore the collisions which are OK (between successive parts of your kinematic chain). 

I could try to add that feature to the robot class of GPU Voxels.

Cheers,
Andreas

bsaund, 2018-03-14:
I will need self-collision checking eventually, and preferably in gpu_voxels rather than duplicating work in a different architecture.
I'd be happy if this functionally appeared.

Side note:
Currently my robot is actually stored in a probab map, partially because conversion, insertion, subtraction, etc. between map types is not all implemented. It makes sense to do the self-collision checking in the preferred map type then implement these additional operations across map types.

Squelsh, 2018-03-14:
Hm, the self collision detection can not be implemented in a probabilistic map, if you want to be able to mask out some collision pairs.
But subtraction and other operations could also be implemented with BitvectorVoxels in the future.

Squelsh, 2018-03-22:
I have the implementation finished. Just adding some auto-testcases before I open a pull request.
Hopefully tomorrow.

Squelsh, 2018-03-28:
Sry, tests discovered some weak points I missed previously. Still on it.

Squelsh, 2018-03-30:
Finally...
The implementation can be found here:
This also contains an example and an automated testcase.
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/pull/68

I am happy to read your feedback.

bsaund, 2018-04-03:
Thanks!
 It'll be a bit before I can look at it.

closed 2018-05-01 by cjue