fzi-forschungszentrum-informatik/gpu-voxels #71: test_gpu_voxels_core - 7 tests fail -

r2b0, 2018-05-29:
Hello,
Thank you to the authors for this nice software.

When I run the sample program:
`test_gpu_voxels_core`

It returns:
`7 tests fail!`

Including errors like:
`cuHandleError: a PTX JIT compilation failed in ... /packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp`

**Is this the expected output for the Master branch?  or is my setup broken?**

**My setup is:**
Ubuntu 14.04.5
Cuda 8 GA2
NVidia GeForce GTX 860M, Kepler, 2GB memory, Capability 3.0
Compiled using ICMAKER_CUDA_COMPUTE_VERSION 30

I also tried a different computer with a
NVidia Quadro M1000M, Maxwell, 2GB memory, Capability 5.0
Compiled using ICMAKER_CUDA_COMPUTE_VERSION 35

Thanks for your help.

Here is the complete output:

```
===========================================================
=      Please run this Benchmark at least two times       =
=   to avoid measuring just in time compilation effects   =
===========================================================
Running 49 test cases...
WARNING: Logging Instance is null, did you initialize the logging framework?
You should initialize the logging framework at the beginning of your program. This will also enable setting the log level on the command line.
<2018-05-28 20:21:04.122> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 74.
<2018-05-28 20:21:04.221> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 75.
<2018-05-28 20:21:04.320> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 78.
<2018-05-28 20:21:04.421> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 81.
<2018-05-28 20:21:04.521> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 99.
<2018-05-28 20:21:04.620> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 101.
<2018-05-28 20:21:04.720> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 106.
<2018-05-28 20:21:04.823> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 109.
<2018-05-28 20:21:04.922> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 240.
unknown location(0): fatal error in "distance_correctness": std::runtime_error: function_attributes(): after cudaFuncGetAttributes: a PTX JIT compilation failed
/home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/test/testing_cudaMath.cu(191): last checkpoint
<2018-05-28 20:21:05.122> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 74.
<2018-05-28 20:21:05.223> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 75.
<2018-05-28 20:21:05.320> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 78.
<2018-05-28 20:21:05.418> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 81.
<2018-05-28 20:21:05.519> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 99.
<2018-05-28 20:21:05.618> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 101.
<2018-05-28 20:21:05.719> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 106.
<2018-05-28 20:21:05.816> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 109.
<2018-05-28 20:21:05.915> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/voxelmap/TemplateVoxelMap.hpp on line 240.
unknown location(0): fatal error in "distance_extraction": std::runtime_error: function_attributes(): after cudaFuncGetAttributes: a PTX JIT compilation failed
/home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/test/testing_cudaMath.cu(191): last checkpoint

insertTest()
create octree....
<2018-05-28 20:21:06.019> CudaLog(Error)::cuCheckForError: a PTX JIT compilation failed(78) in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/test/../octree/test/Tests.cu on line 495.
<2018-05-28 20:21:06.119> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp on line 328.
<2018-05-28 20:21:06.218> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp on line 329.
<2018-05-28 20:21:06.318> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp on line 330.
<2018-05-28 20:21:06.418> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp on line 331.
unknown location(0): fatal error in "insert_voxels": std::bad_alloc: std::bad_alloc: a PTX JIT compilation failed
/home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/test/testing_octree.cu(56): last checkpoint
Running on GPU 0 (GeForce GTX 860M)
<2018-05-28 20:21:06.620> CudaLog(Error)::cuHandleError: a PTX JIT compilation failed in /home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/octree/test/Helper.cpp on line 64.
[... 134 lines cut ...]
runtime_median: 5889.43 ms
runtime_min: 5889.43 ms
runtime_max: 5889.43 ms

Opened log file GPUVoxelsBenchmarkProtocol_2018-05-29_00_21.txt<2018-05-28 20:21:09.799> Performance(Info)::print: 
########## Begin of Summary for prefix voxellists ##########

<2018-05-28 20:21:09.799> Performance(Info)::print: 
########## Begin of Summary for prefix voxelmap ##########

Found 1 devices.
Dev 0 = Device Information of GPU 0
Model: GeForce GTX 860M
Multi Processor Count: 6
Global Memory: 1997.88 MB
Shared Memory per Block: 49152 Shared Memory per Multi Processor: 49152
Max Threads per Block 1024 Max Threads per Multi Processor: 2048
Registers per Block: 65536 Registers per Multi Processor: 65536
Max grid dimensions: [ 2147483647, 65535, 65535 ]
Max Block dimension: [ 1024, 1024, 64 ]
Warp Size: 32

*** 7 failures detected in test suite "Master Test Suite"

```

cjue, 2018-05-29:
Hi, 

thanks for your feedback!

Which NVIDIA driver version are you using? "nvidia-smi" will print it.

I recently found similar problems with Cuda 8 and GPU Voxels 1.2.0 and driver versions 384 onward, while everything works with the older driver version 375.66, or with Cuda 9.x

We are currently debugging the issue.

r2b0, 2018-05-30:
I tried disabling the JIT compilation by changing
`packages/gpu_voxels/CMakeLists.txt`
to use these settings:
`-gencode arch=compute_35,code=sm_35`

The compilation fails again with the same error (or at least the first error), but now it gives more information:
```
[ 77%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir/test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o
ptxas fatal   : Unresolved extern function '_ZN10gpu_voxels13DistanceVoxelC1Ev'
CMake Error at test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o.cmake:264 (message):
  Error generating file
  /home/home/username/gpu-voxels-master/build/packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir//./test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o
```

Detailed output below:
```
<snip>

Linking CXX shared library ../../../../../../lib/libgpu_voxels_urdf_robot.so
[ 67%] Built target gpu_voxels_urdf_robot
Linking CXX shared library ../../../../lib/libgpu_voxels.so
[ 67%] Built target gpu_voxels
[ 68%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap_tests_CUDA_TARGET.dir/gpu_voxels_voxelmap_tests_CUDA_TARGET_generated_Tests.cu.o
[ 68%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap_tests_CUDA_TARGET.dir/kernels/gpu_voxels_voxelmap_tests_CUDA_TARGET_generated_VoxelMapOperations.cu.o
[ 68%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap_tests_CUDA_TARGET.dir/kernels/gpu_voxels_voxelmap_tests_CUDA_TARGET_generated_VoxelMapTests.cu.o
Linking CXX shared library ../../../../../lib/libgpu_voxels_voxelmap_tests.so
[ 68%] Built target gpu_voxels_voxelmap_tests
Linking CXX executable ../../../../../../bin/octree_bench
[ 72%] Built target octree_bench
Linking CXX executable ../../../../../../bin/octree_provider
[ 75%] Built target octree_provider
[ 76%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers_highlevel/CMakeFiles/gpu_voxels_distance_map_converter_CUDA_TARGET.dir/gpu_voxels_distance_map_converter_CUDA_TARGET_generated_DistanceMapConverter.cu.o
Linking CXX shared library ../../../../../lib/libgpu_voxels_distance_map_converter.so
[ 76%] Built target gpu_voxels_distance_map_converter
[ 76%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir/__/octree/test/test_gpu_voxels_core_CUDA_TARGET_generated_Tests.cu.o
[ 76%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir/test_gpu_voxels_core_CUDA_TARGET_generated_testing_bitvector.cu.o
[ 77%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir/test_gpu_voxels_core_CUDA_TARGET_generated_testing_cudaMath.cu.o
/home/home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/test/testing_cudaMath.cu(140): warning: controlling expression is constant

/home/home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/test/testing_cudaMath.cu(140): warning: controlling expression is constant

[ 77%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir/test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o
ptxas fatal   : Unresolved extern function '_ZN10gpu_voxels13DistanceVoxelC1Ev'
CMake Error at test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o.cmake:264 (message):
  Error generating file
  /home/home/username/gpu-voxels-master/build/packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir//./test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o

make[2]: *** [packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir/test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o] Error 1
make[1]: *** [packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core.dir/all] Error 2
make: *** [all] Error 2
```

r2b0, 2018-05-30:
Reverting from 384.11 to 375.82 was non-trivial for me, so here's some instructions:

In Ubuntu 14.04, you can install NVidia drivers using the 'graphics-drivers' PPA  
(http://www.linuxandubuntu.com/home/how-to-install-latest-nvidia-drivers-in-linux)  
However, Aptitude shows that the package 'nvidia-375' is just an alias (transitional package) for the 384 driver.

Maybe you can use the 'nvidia-340' driver but it's not compatible with my graphics card.  
See here:  
https://us.download.nvidia.com/XFree86/Linux-x86/340.106/README/supportedchips.html

Therefore I installed the 375.82 driver the old-fashioned way. Manually.  
More info here:  
https://askubuntu.com/questions/481414/install-nvidia-driver-instead-of-nouveau

Download the NVidia driver from the archive:
http://www.nvidia.com/object/linux-amd64-display-archive.html

Purge all the NVidia packages and go back to a fresh start.  
Blacklist the nouveau kernel module, etc.  
Kill your display manager and install the driver from a terminal using the .run file.

Verify the installation:  
`nvidia-smi`

Remove any previous PPA or debs for CUDA 8 and the patch.  
The .deb installer keeps trying to install the 384 driver!

Instead, download the .run file installer for CUDA 8 GA2.

I noticed the filename says "version 8.0.61" and driver "375.26".  
It seems to work ok with the newer 375.xx  
When you run the installer, it will ask if you want to install that version of the driver.  
Choose "No".

To verify the installation, compile the sample code.  
All the programs should successfully compile. Previously when I had driver 384.xx, the compilation failed.  
Run this:  
`./deviceQuery`

One thing to be careful of, is that when I run  
`sudo apt-get update`
It tells me that all the CUDA 8 packages were installed but are no longer required.
It suggests that I remove them!

Re-compile GPU-Voxels.

closed 2018-05-30 by r2b0