fzi-forschungszentrum-informatik/gpu-voxels #72: Visualizer: Couldn't open the shared memory segment of Visualizer! - gpu_voxels_visualizer

r2b0, 2018-05-29:
Hello again,
When I run the Visualizer it appears to work 90% okay, but there's an error message in the terminal:
`Couldn't open the shared memory segment of Visualizer!`

**What does this error mean please?**

Sometimes there are some strange artifacts in the Visualizer, like things appearing in the wrong place.
Or if I use the mouse scroll, the voxels are drawn in the wrong place.
The boxes (probabilistic voxmap) are flashing two shades of blue, but I guess this is normal.
It would be helpful to add some more images to your documentation, because the Visualizer output looks completely different to all the figures on the website or in your publications.

Steps to reproduce:
```
./collisions
./gpu_voxels_visualizer 
```

Here is the output:
```
gpu-voxels-master/build/bin$ ./gpu_voxels_visualizer 
<2018-05-28 20:36:09.116> Visualization(Info)::main: Starting the gpu_voxels Visualizer.
<2018-05-28 20:36:09.116> Visualization(Warning) Visualizer::initalizeVisualizer: Couldn't open the shared memory segment of Visualizer!
CONFIG WARNING: The configuration framework is already initialized!
<2018-05-28 20:36:09.117> Visualization(Warning) XMLInterpreter::getUnitScale: No unit scale defined. Using 1 cm instead.
<2018-05-28 20:36:09.280> Visualization(Info) Visualizer::initGL: Using GPU Device 0: "GeForce GTX 860M" with compute capability 3.0
<2018-05-28 20:36:09.471> Visualization(Info) Visualizer::initGL: Using OpenGL Version: 4.5.0 NVIDIA 384.111
<2018-05-28 20:36:09.471> Visualization(Info) Visualizer::initGL: Environment variables not set (MCAHOME)! The default path is used instead.
<2018-05-28 20:36:09.472> Visualization(Info) Visualizer::initGL: Search path of the shaders is: /home/username/gpu-voxels-master/build/shader
<2018-05-28 20:36:09.475> Visualization(Info)::main: Number of voxel maps that will be drawn: 1
```

**My setup is:**
Ubuntu 14.04.5
Cuda 8 GA2
NVidia GeForce GTX 860M, Kepler, 2GB memory, Capability 3.0
Compiled using ICMAKER_CUDA_COMPUTE_VERSION 30

I also tried a different computer with a
NVidia Quadro M1000M, Maxwell, 2GB memory, Capability 5.0
Compiled using ICMAKER_CUDA_COMPUTE_VERSION 35

cjue, 2018-05-29:
Hi,
thanks again for the detailed feedback! We are currently working on improving the examples and their documentation.

> if I use the mouse scroll, the voxels are drawn in the wrong place.

If you simply scroll, you change the "super voxel size", thereby changing the rendering resolution for the obstacles voxelmap. The combination for zooming in and out is Ctrl + Alt + MOUSE_WHEEL. While thinking about it, it might be a good idea to switch these for the next release. All key-bindings are printed to the console on pressing "h" in the visualizer.

> The boxes (probabilistic voxmap) are flashing two shades of blue, but I guess this is normal.

Yes, this is an artifact of the "insertBoxIntoMap" function, which generates a pointcloud that will then be inserted. Because of discretization artifacts, some of the obstacle voxels are hit by multiple insertions and colored differently.

> the Visualizer output looks completely different to all the figures on the website

The color schemes can be changed by providing a different visualizer configuration. We will include examples for this in the documentation update.

>> Couldn't open the shared memory segment of Visualizer!
>
> What does this error mean please?

This warning message is needlessly confusing, as everything works as intended. I will downgrade the logging level so it's not shown by default for the next release. I also fixed the Config double initialization error.

r2b0, 2018-05-30:
Thanks for your quick reply.

**Regarding the visualizations:**

Were these images made from the Visualizer or from ROS?
http://www.gpu-voxels.org/wp-content/uploads/2014/03/filmstrip-1500x600.jpg
http://www.gpu-voxels.org/wp-content/uploads/2014/03/galerie_3.jpg

**Regarding the scroll wheel and "super voxel size":**
I was comparing the output of two examples:

./robot_vs_environment
When you scroll to increase the voxel size, the voxels get bigger and they are centered in same  position. This looks correct.

./collisions
When you scroll, the voxels of the cubes seem to shift to the side as they get bigger. This doesn't look right.

Perhaps I need to get some screenshots or code up another example.

cjue, 2018-05-30:
Hi,

these screenshots were both made by Andreas, maybe he will join the conversation later.

Both of the screenshots you mentioned appear to be from RViz, based on the rendering of the robot model and the "interactive markers with arrows".

Regarding the supervoxels:
To my eye they work as expected in the collisions example: The super voxels are aligned to a voxelgrid with identical origin, although with larger voxel size. So if the super-voxels appear to grow centered around the original voxel, that is merely a coincidence: the supervoxel would be drawn in the same position if the original voxel  had been somewhere else within that supervoxel's volume.

cjue, 2018-08-28:
The confusing log messages should be fixed by Release 1.2.1 feb56b3

closed 2018-08-28 by cjue