fzi-forschungszentrum-informatik/gpu-voxels #73: Can't compile without JIT (Just In Time)

r2b0, 2018-05-30:
I thought I should make this a separate issue, because I didn't solve it yet.

By default, the JIT compilation is disabled.
See this file:  
`packages/gpu_voxels/CMakeLists.txt`

This line:
`SET(ICMAKER_CUDA_COMPUTE_VERSION 35)`
makes it compile using:
`-gencode arch=compute_35,code=compute_35`
and then it does the JIT compile at runtime, which causes a delay.

I tried using these settings:
`-gencode arch=compute_30,code=sm_30`
or
`-gencode arch=compute_35,code=sm_35`
or
`-gencode arch=compute_50,code=sm_50`
but now the compilation fails with an error:
```
[ 77%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir/test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o
ptxas fatal   : Unresolved extern function '_ZN10gpu_voxels13DistanceVoxelC1Ev'
CMake Error at test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o.cmake:264 (message):
  Error generating file
  /home/home/username/gpu-voxels-master/build/packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir//./test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o
```

.

Detailed output below:
```
<snip>

Linking CXX shared library ../../../../../../lib/libgpu_voxels_urdf_robot.so
[ 67%] Built target gpu_voxels_urdf_robot
Linking CXX shared library ../../../../lib/libgpu_voxels.so
[ 67%] Built target gpu_voxels
[ 68%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap_tests_CUDA_TARGET.dir/gpu_voxels_voxelmap_tests_CUDA_TARGET_generated_Tests.cu.o
[ 68%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap_tests_CUDA_TARGET.dir/kernels/gpu_voxels_voxelmap_tests_CUDA_TARGET_generated_VoxelMapOperations.cu.o
[ 68%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap_tests_CUDA_TARGET.dir/kernels/gpu_voxels_voxelmap_tests_CUDA_TARGET_generated_VoxelMapTests.cu.o
Linking CXX shared library ../../../../../lib/libgpu_voxels_voxelmap_tests.so
[ 68%] Built target gpu_voxels_voxelmap_tests
Linking CXX executable ../../../../../../bin/octree_bench
[ 72%] Built target octree_bench
Linking CXX executable ../../../../../../bin/octree_provider
[ 75%] Built target octree_provider
[ 76%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers_highlevel/CMakeFiles/gpu_voxels_distance_map_converter_CUDA_TARGET.dir/gpu_voxels_distance_map_converter_CUDA_TARGET_generated_DistanceMapConverter.cu.o
Linking CXX shared library ../../../../../lib/libgpu_voxels_distance_map_converter.so
[ 76%] Built target gpu_voxels_distance_map_converter
[ 76%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir/__/octree/test/test_gpu_voxels_core_CUDA_TARGET_generated_Tests.cu.o
[ 76%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir/test_gpu_voxels_core_CUDA_TARGET_generated_testing_bitvector.cu.o
[ 77%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir/test_gpu_voxels_core_CUDA_TARGET_generated_testing_cudaMath.cu.o
/home/home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/test/testing_cudaMath.cu(140): warning: controlling expression is constant

/home/home/username/gpu-voxels-master/packages/gpu_voxels/src/gpu_voxels/test/testing_cudaMath.cu(140): warning: controlling expression is constant

[ 77%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir/test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o
ptxas fatal   : Unresolved extern function '_ZN10gpu_voxels13DistanceVoxelC1Ev'
CMake Error at test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o.cmake:264 (message):
  Error generating file
  /home/home/username/gpu-voxels-master/build/packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir//./test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o

make[2]: *** [packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core_CUDA_TARGET.dir/test_gpu_voxels_core_CUDA_TARGET_generated_testing_distance.cu.o] Error 1
make[1]: *** [packages/gpu_voxels/src/gpu_voxels/test/CMakeFiles/test_gpu_voxels_core.dir/all] Error 2
make: *** [all] Error 2
```

cjue, 2018-05-30:
Hi r2b0,

first off, because of JIT caching, the delay luckily only occurs on the first run after a new build.

I know about the compile error with `code=sm_*`. I would be grateful for any pointers as to how this can be avoided - my fear is that this would entail significant changes to the build process.

r2b0, 2018-06-08:
Hello,
The problem is in `src/gpu_voxels/test/testing_distance.cu`  
The second test case uses the class DistanceVoxel, but the compiler doesn't know about that.  

Just add this header:  
`#include <gpu_voxels/voxel/DistanceVoxel.hpp>  `
now you can compile with `code=sm_35`

I don't know if this is the correct way to solve this bug, but it works.  Also, I saw that other parts of the code include the .hpp file, not the .h file.

Pull request:  https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/pull/74

cjue, 2018-06-08:
Thanks for the input!

I'll run some tests with your changes. This could help with some other issues as well.

cjue, 2018-08-28:
Fixed by Release 1.2.1 feb56b3

Thanks for your pull request r2b0! The pull request includes your proposed fix.

closed 2018-08-28 by cjue