fzi-forschungszentrum-informatik/gpu-voxels #75: gpu_voxels_helpers Doesn't build

trent-weiss, 2018-06-19:
Hello!

I am attempting to build on Ubuntu 16.04 with ROS Kinetic installed.

The CMake configure step seems to run just fine.

But when I run make, I get the following error:

> [  6%] Built target icl_core
[ 10%] Built target icl_core_config
[ 17%] Built target icl_core_logging
[ 18%] Built target icl_core_plugin
[ 24%] Built target icl_core_thread
[ 25%] Built target icl_core_dispatch
[ 26%] Built target icl_core_performance_monitor
[ 26%] Built target icl_core_performance_monitor_example
[ 28%] Built target uls
[ 29%] Built target test_icl_core_convert
[ 29%] Built target test_icl_core_config
[ 30%] Built target test_icl_core_logging
[ 34%] Built target ts_icl_core
[ 35%] Built target ts_icl_core_config
[ 37%] Built target ts_icl_core_thread
[ 38%] Built target ts_icl_core_crypt
[ 41%] Built target gpu_voxels_logging
[ 41%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/kernels/gpu_voxels_helpers_CUDA_TARGET_generated_HelperOperations.cu.o
[ 42%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/gpu_voxels_helpers_CUDA_TARGET_generated_cuda_handling.cu.o
[ 42%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/gpu_voxels_helpers_CUDA_TARGET_generated_PointCloud.cu.o
[ 42%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/gpu_voxels_helpers_CUDA_TARGET_generated_MetaPointCloud.cu.o
[ 43%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/kernels/gpu_voxels_helpers_CUDA_TARGET_generated_MetaPointCloudOperations.cu.o
[ 43%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/gpu_voxels_helpers_CUDA_TARGET_generated_cuda_handling.cu.o
[ 43%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/gpu_voxels_helpers_CUDA_TARGET_generated_PointCloud.cu.o
[ 43%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/gpu_voxels_helpers_CUDA_TARGET_generated_MetaPointCloud.cu.o
[ 43%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/kernels/gpu_voxels_helpers_CUDA_TARGET_generated_MetaPointCloudOperations.cu.o
[ 43%] Building NVCC (Device) object packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/kernels/gpu_voxels_helpers_CUDA_TARGET_generated_HelperOperations.cu.o
[ 43%] Linking CXX shared library ../../../../../lib/libgpu_voxels_helpers.so
c++: error: CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/gpu_voxels_helpers_CUDA_TARGET_generated_cuda_handling.cu.o: No such file or directory
c++: error: CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/gpu_voxels_helpers_CUDA_TARGET_generated_PointCloud.cu.o: No such file or directory
c++: error: CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/gpu_voxels_helpers_CUDA_TARGET_generated_MetaPointCloud.cu.o: No such file or directory
c++: error: CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/kernels/gpu_voxels_helpers_CUDA_TARGET_generated_MetaPointCloudOperations.cu.o: No such file or directory
c++: error: CMakeFiles/gpu_voxels_helpers_CUDA_TARGET.dir/kernels/gpu_voxels_helpers_CUDA_TARGET_generated_HelperOperations.cu.o: No such file or directory
packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers.dir/build.make:601: recipe for target 'lib/libgpu_voxels_helpers.so' failed
make[2]: *** [lib/libgpu_voxels_helpers.so] Error 1
CMakeFiles/Makefile2:4298: recipe for target 'packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers.dir/all' failed
make[1]: *** [packages/gpu_voxels/src/gpu_voxels/helpers/CMakeFiles/gpu_voxels_helpers.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

cjue, 2018-06-19:
Hi Trent,

could you please report your CMake version?

```
$ cmake --version
cmake version 3.5.1
```

trent-weiss, 2018-06-19:
>

cjue, 2018-06-19:
Hi Trent, 

this is caused by an issue with CMake newer than 3.5, as described in issue #58.

The quick workaround is to use CMake 3.5, the default for Ubuntu 16.04.

We have found a partial solution for this and will include it in the next release, as soon as we are sure it doesn't break anything. #

cjue, 2018-06-19:
Duplicate of #58

closed 2018-06-19 by cjue

trent-weiss, 2018-06-19:
yepp. I can verify that running "/usr/bin/cmake .." from the build directory (forcing bash to use the system cmake) resolves the issue.

Thanks.