fzi-forschungszentrum-informatik/gpu-voxels #76: Const Correctness on Interfaces

oscarmendezm, 2018-07-27:
In PointCloud.cu and PointCloud.h, should the function PointCloud::transform(...) be const?

Like so:

-  void transform(const Matrix4f* transform, PointCloud* transformed_cloud);
+  void transform(const Matrix4f* transform, PointCloud* transformed_cloud) const;

Squelsh, 2018-07-28:
In general you are right. The problem here is the m_transformation_dev variable which is a class variable that is used. This variable could be marked as `mutable` nevertheless so that the function could be `const`, as you suggest. Perhaps @cjue  wants to change this in the next release?

I am sure, we could fine more candidates, which could be `const`, apart of a temporary device variable.

cjue, 2018-08-01:
Thanks for the comments. I'll address PointCloud const correctness in an upcoming update.