fzi-forschungszentrum-informatik/gpu-voxels #77: Problem to compile code. Not found openni_grabber.h

ilia-sheremet, 2018-07-27:
Hello there,

I'm having issued compiling the code with following error
`Kinect.h:28:35: fatal error: pcl/io/openni_grabber.h: No such file or directory`

I have PCL and OpenNI installed. The file **openni_grabber.h** is present in the file system under following path:
`/usr/include/pcl-1.7/pcl/io/openni_grabber.h`

Here is the output of **$PATH**:
`/opt/ros/kinetic/bin:/home/ilia-dev/bin:/home/ilia-dev/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/include`

Output of **uname -a**:
`Linux pdgcs01-X555LJ 4.15.0-29-generic #31~16.04.1-Ubuntu SMP Wed Jul 18 08:54:04 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

Output of **nvcc --version**:
`nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2015 NVIDIA Corporation
Built on Tue_Aug_11_14:27:32_CDT_2015
Cuda compilation tools, release 7.5, V7.5.17`

[Link](https://paste.ofcode.org/KpS6AciP9WjdEPzuDT66Y9) to the output of **cmake** command.

Could someone please tell me the versions being used for the repo dependencies? Especially PCL and OpenNI. Maybe someone has script to install those dependencies?

Thank you!

cjue, 2018-08-01:
Hi Ilia,
thank you for your report.!

I see your are using ROS kinetic, Cuda 7.5 and libopenni 1.5.4
Are you using Ubuntu 16.04 with libpcl version 1.7.2?

ilia-sheremet, 2018-08-03:
@cjue , I had actually both version installed 1.7.2 from repos, and 1.8.1 from source. 
@Squelsh , I made a mistake to look for the file **openni_grabber.h** which was present only in 1.7.2, so I assumed the 1.8.1 wasn't installed properly.

Now I removed manually all the pcl files and installed 1.7.2 from repo again and now everything compiles properly.

Here is the output for **sudo apt list --installed | grep pcl**, in case someone will have similar version issues:
`libdapclient6v5/xenial,now 3.15.1-7 amd64 [installed,automatic]
libpcl-apps1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-common1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-dev/xenial,now 1.7.2-14build1 amd64 [installed]
libpcl-features1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-filters1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-io1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-kdtree1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-keypoints1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-octree1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-outofcore1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-people1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-recognition1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-registration1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-sample-consensus1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-search1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-segmentation1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-surface1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-tracking1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl-visualization1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
libpcl1.7/xenial,now 1.7.2-14build1 amd64 [installed,automatic]
ros-kinetic-pcl-msgs/xenial,now 0.2.0-0xenial-20180516-152510-0800 amd64 [installed]
`

@cjue @Squelsh , I think it would be a great idea to include versions for dependencies packages in Readme and I could make PR with those changes. But I could name the versions that I compiled with on my PC and I don't know how much backwards compatibility work with other versions of PCL, OpenNi, Boost and tinyXML

cjue, 2018-08-03:
Thanks for the offer, but are you sure that PCL 1.8.1 is acutally missing that include?

There is a file of that name in 1.8.1 on Github: https://github.com/PointCloudLibrary/pcl/tree/pcl-1.8.1/io/include/pcl/io

I am interested in making sure the library is comaptivle with PCL 1.8.1.

We are currently also doing some testing regarding Boost to ensure compatibility with all releases starting from Ubuntu 14.04.

ilia-sheremet, 2018-08-06:
@cjue I think so, I ran **find**  through out the system, searching for the file. But I am not 100% sure.

BTW do you know if your library works with CUDA 8.0? And if not do you have plans for importing it?

cjue, 2018-08-21:
Hi Ilia,

GPU-Voxels is Cuda 8.0 compatible, with a slight caveat: 

`Code compiled with Cuda 8.0 works fine with older GPU drivers such as 375.66, but there are runtime errors with driver 384.111 and newer ("PTX JIT compilation failed").`

RickHutten, 2019-03-06:
I had the same problem as Ilia. I have installed both PCL 1.7.2 and 1.9.1. I tried uninstalling 1.7.2 but still had the same problem. I found that the file openni_grabber.h was not present in the pcl 1.9.1 installation. When rebuilding PCL 1.9.1 I noticed with running ccmake that the option **WITH_OPENNI** was set to "OFF". Turning it on and rebuilding PCL did the trick for me. Hope this might help.

cjue, 2019-03-06:
@Rick and Ilia: do you know of a way to check whether a PCL installation has been compiled with activated WITH_OPENNI setting?

If there is something like that, I could improve our cmake detection of PCL&Openni. We only need the OpenNI support in PCL in order to directly access cameras, through the "helpers/Kinect" class.

cjue, 2019-03-06:
To answer myself: We could just look for "pcl/io/openni_grabber.h" within PCL_INCLUDE_DIRS and make that a condition for the optional helpers/Kinect target.

I'll reopen this as a reminder for myself.

rick, 2019-03-08:
[image]