fzi-forschungszentrum-informatik/gpu-voxels #78: 268 Fails - Test_gpu_voxels_core

AlejoDiaz49, 2018-07-27:
Hello,

I just start working with this repository and I am having some problems. When I run the example program
`test_gpu_voxels_core`
It detects 268 failures, but the most recurrent error even when I run some other examples is
`CudaLog(Error)::cuCheckForError: too many resources requested for launch(7) in /home/nvidia/Repos/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/voxelmap/BitVoxelMap.hpp`
I thought it was some problem of the memory being full, but everything seems fine.

My Setup is:
- Jetson TX2
- Ubuntu 16.04 (Working without GUI)
- CUDA 9.0.252
- TinyXML 2.6.2v5
- Boost 1.58.0
- PCL 1.81

I am working without OpenNI, ROS and the visualization packages, so when I build this is the message
`------------------------ GPU Voxels configuration ------------------------
--
-- [OK]      Building GPU-Voxels. Found CUDA.
-- [WARNING] Building GPU-Voxels without visualization because the following packages are missing:
--     Glew
--     GLM
-- [WARNING] Building GPU-Voxels without ROS connections. ROS not found.
-- [WARNING] Building GPU-Voxels without URDF support. Could not find the following packages:
--     urdfdom
--     orocos_kdl
--     kdl_parser
--     ROS
-- [WARNING] Building GPU-Voxels without Kinect support. OpenNI not found.
-- [OK]      Building GPU-Voxels with PCL interfaces. PCL found.
--
-- -------------------- END of GPU Voxels configuration ---------------------`

I understand the errors about visualization, but I don't know if I its normal to have so many errors in the test. Here I attached the result of it.

[GPUVoxelsBenchmarkProtocol_2018-07-27_15_18.txt](https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/files/2236369/GPUVoxelsBenchmarkProtocol_2018-07-27_15_18.txt)

Thanks in advanced

cjue, 2018-07-28:
Hi Alejo,

this is probably related to the "--maxrregcount" setting in packages/GPU-Voxels/CMakeLists.txt

As mentioned in the comment, this limit needs to be lowered for Jetson boards.

Does this fix your issue?

AlejoDiaz49, 2018-07-31:
Hi, thanks for the reply

Yes it works, now it doesn't detect errors. Only two more things, at the beginning this message appears

`WARNING: Logging Instance is null, did you initialize the logging framework?`
`You should initialize the logging framework at the beginning of your program. This will also enable setting the log level on the command line.`

And also in two tests a message indicating an error appears

`checking unknown/free voxel...`
`Error! Unknown voxel with ID 1320 not found in octree.`
`<2018-07-31 08:59:42.648> OctreeLog(Info)::checkTree: checking tree . . .`
`ERROR InnerNode on level 3 for voxel with id 1024 has wrong status flags`
`<2018-07-31 08:59:42.650> OctreeLog(Error)::checkTree: ##### ERROR checkTree() FAILED #####`
`##### insertTest() finished with ERRORS #####`

`checking unknown/free voxel...`
`Error! Unknown voxel with ID 88 not found in octree.`
`<2018-07-31 09:00:12.443> OctreeLog(Info)::checkTree: checking tree . . .`
`ERROR InnerNode on level 2 for voxel with id 64 has wrong status flags`
`<2018-07-31 09:00:12.445> OctreeLog(Error)::checkTree: ##### ERROR checkTree() FAILED #####`
`##### insertTest() finished with ERRORS #####`

Is all of these normal?

Thanks in advanced