fzi-forschungszentrum-informatik/gpu-voxels #79: Visualization errors launching the examples

AlejoDiaz49, 2018-08-01:
Hello,

I am having errors in some of the examples and I the visualization is not working. All the tests are fine, they don't show any error, except **test_icl_core_looging** that shows this message

`Main(Info)::main: Creating logging thread.`
`Main(Info)::main: Starting logging thread.`
`ThreadLog(Error) LoggingThread::run: Thread Error`
`ThreadLog(Warning) LoggingThread::run: Thread Warning`
`ThreadLog(Info) LoggingThread::run: Thread Info`
`ThreadLog(Info) LoggingThread::run: Thread Loop 0`
`Main(Info)::main: Main Loop 0`

I am not using ROS and OpenNI. My setup is:
 - Jestson TX2
 - Ubuntu 16.04
 - CUDA 9.0.252
 - TynyXML 2.6.2v5
 - Boost 1.58.0
 - PCL 1.8.1
 - GLUT 2.8.1-2
 - GLEW 1.13.0-2
 - GLM 0.9.7.2-1

The main problem is this error message I get with almost all the examples

`CudaLog(Error)::cuHandleError: operation not supported in /home/nvidia/Repos/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/vis_interface/VisVoxelMap.cpp on line 86.`

Also in the examples **collision**, **counting_sandbox** this error appears

`CudaLog(Info)::cuTestAndInitDevice: Running on GPU 0 (NVIDIA Tegra X2)`
`Name of shared buffer swapped: voxel_list_buffer_swapped_0.`
`CudaLog(Error)::cuHandleError: operation not supported in /home/nvidia/Repos/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/vis_interface/VisTemplateVoxelList.hpp on line 112.`
`CUDA error 71 [/usr/local/cuda/include/thrust/system/cuda/detail/parallel_for.h, 143]: operation not supported`
`terminate called after throwing an instance of 'thrust::system::system_error'`
`  what():  parallel_for failed: operation not supported`
`Aborted (core dumped)`

And in **gpu_voxels_visualizer** this one

`SharedMemManager(Error) SharedMemoryManagerPrimitiveArrays::getPrimitivePositions: Primitive Arrays Handler could not be opened! Error was 71`
`Visualization(Error) Visualizer::drawPrimitivesFromSharedMem: It was not possible to load primitive data from shared memory.`

I also got the sames errors after changing the bug lines in the **func_common.inl** file of GLM. Is a problem with Octrees, for not been using OpenNI or ROS or for the versions of GLEW, GLM and GLUT?

Squelsh, 2018-08-01:
I think, the logging output is not really an error, but a demonstration of the various log-stream severities.

independent of that: You can not use the visualizer on a Tegra board, as this hardware does not support shared memory as we use it (as the RAM is already shared by the hardware anyway).

Perhaps we once offer a special version of GPU Voxels only for embedded ARM devices. Up till then, you can only try to serialize your data to a file, deserialize it on a X86 platform, and visualize it there.

As an alternative, if you have a bit of experience in CUDA coding, you could change the SharedMemory implementation and adapt it to the ARM board. It should be doable with very little lines of code at some central files.

Please look at  https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/issues/65 for more details.

cjue, 2018-08-02:
Hi Alejandro,

as mentioned by Andreas,  the test_icl_core_looging output works as expected and the visualizer errors are a result of the lacking support for the shared memory primitives GPU-Voxels uses on x86 PCs.

It would require changes to the way data is exchanged between GPU-Voxels programs and the OpenGL visualizer, which runs in a separate process.

I will have another look into issue #65. If using the visualizer directly on the Jetson board is important for your use-case, I can help you look into the required changes to the shared memory calls.

Closed as duplicate of #65

closed 2018-08-02 by cjue

AlejoDiaz49, 2018-08-06:
Hi Andreas and Christian

I can avoid using the visualizer in the Jetson, and just test the programs in another computer. 
In any case I will probably try to implement the serialization of the data.

Thanks for the replies