fzi-forschungszentrum-informatik/gpu-voxels #82: Multiple initializazion of GPU-Voxels

giorgionicola, 2018-09-11:
Hi,
I have an application where it would be very useful the possibility to initialize 2 different "collision scenes" with different resolutions (= voxels length), one for the collisions between the two robots bodies and another to check the collision between the two robots' tools.
Although at the moment this is impossible since there some static elements in the libraries. Does exist any method to overcome this limitation? Will it be possible in future to initialize multiple times GPU-Voxels? Those static elements are absolutely necessary?
thanks

cjue, 2018-09-12:
Hi,

it is already possible to directly construct a GpuVoxelsMap with custom parameters, using the same API as the addMap function:

https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/gpu_voxels/GpuVoxels.cpp#L189

`voxellist::BitVectorVoxelList* orig_list = new voxellist::BitVectorVoxelList(m_dim, m_voxel_side_length, MT_BITVECTOR_VOXELLIST);`

If you use this directly, your map will not be a "manged map" and you cannot interact with it through functions that take a map name.

Depending on your use-case this can still be useful, e.g. when performing collision checks:

https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/gpu_voxels/test/testing_voxellist.cu#L44

Does this match your use-case?

giorgionicola, 2018-09-12:
Ok but it is not clear to how I can interact with this new custom maps created.
In my application the 2 robots tools have 2 urdf files and at every collision check I update their position with the command "setRobotConfiguration" giving the cartesian coordinates (x,y,z,roll,pitch,yaw) and I use the commend "insertRobotIntoMap" . Although as you said I should not be able to use it since it requires the "map_name".

So if i have understood correctly in my case it should work in this way:

1) I initilaize my GPUVoxelsSharedPtr normally and I use the command "addRobot" for all 4 (2 robots and 2 tools) so i can parse my urdfs, but I use gvl->addMap only for the robots.
2) I add the tools maps as you have shown
3)Then the commend "setRobotConfiguration" to update tools position does not change
4)Then instead of using "insertRobotIntoMap" i do

    ManagedRobotsIterator rob_it = m_managed_robots.find(robot_name);
    insertMetaPointCloud(*rob_it->second->getTransformedClouds(), voxel_meaning);

5) Then i should be able to do the collision checks similarly to the test file

Am I correct?

giorgionicola, 2018-09-19:
I have tried to implement what I have written in the previous post, although i was not successful since `m_managed_robots` is a private variable.

Do you have any suggestions @cjue ?

Squelsh, 2018-09-29:
Hi!
I try to help you out with Christians suggestions. First of all: If you create maps in the way that Christian suggested, those will NOT be manages by GPU-Voxels itself. You have to manage them, i.e. interact with the pointers to them directly.

GPU-Voxels offers two layers of interfaces: The high-level interface, which you call directly on the gvl Pointer. This uses the managed maps and managed robots to work with them. Internally it calls the lower level functions of the map-interfaces.

This is what you have to work with. Do not let gvl create the maps for you, but create them manually. So you can initialize them with different resolutions. Afterwards have a look into the high-level-API implementation and look up the specific functions you need, to e.g. insert a robot into your unmanaged map.

What you will need is this:
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/294e865734d563c7e4a3df8d561e7fc417eefd52/packages/gpu_voxels/src/gpu_voxels/GpuVoxels.cpp#L514

But you call this function directly on your manually created map. 

Hope that helps?

Squelsh, 2018-09-29:
Hi,

almost correct. But instead of interacting with `m_managed_robots` you should use
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/294e865734d563c7e4a3df8d561e7fc417eefd52/packages/gpu_voxels/src/gpu_voxels/GpuVoxels.cpp#L338

On the returned RobotInterfaceSharedPtr you can than call `getTransformedClouds()`

Hope that helps,
Andy

giorgionicola, 2018-10-05:
Hi,
it worked smoothly, thanks

closed 2018-10-19 by cjue