fzi-forschungszentrum-informatik/gpu-voxels #83: gpu-voxels-ROS-workspace: VoxelmapLog(Error)::parallelBanding3D: parallelBanding3D: dimX and dimY must be equal; they also must be divisible by 64

jlllweco, 2018-09-18:
I try to implement gpu-voxel to ROS workspace, and subscribe pointcloud topic `"/zed/point_cloud/cloud_registered"` data from zed-ros-wrapper ZEDmini camera.
But in the code,
`pbaDistanceVoxmap->insertPointCloud(myKinectCloud, eBVM_OCCUPIED);`
 I try to take in real-time point-cloud to calculate pbaDistanceVoxmap.
however, the function:
`cudaMemcpy(&dv, (dvm_thrust_ptr+id).get(), sizeof(DistanceVoxel), cudaMemcpyDeviceToHost);`
reports error:
`VoxelMap::insertPointCloud: You tried to insert points that lie outside the map dimensions!`
`VoxelmapLog(Error)::parallelBanding3D: parallelBanding3D: dimX and dimY must be equal; they also must be divisible by 64.`

May I know what's the unit in '`map damensions'`, it is meter in ZED. The definition code is:
`Vector3ui map_dimensions(256, 256, 256);`
`const float voxel_side_length = 0.01;`

What does the 64 mean in the error code? Do we use the parallelBanding3D function wrongly?

I noticed that m_data is dimension of 640*480; is this a format of depth image instead of point cloud data?

Thank you!
[
[pcl_gpu.tar.gz](https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/files/2392695/pcl_gpu.tar.gz)
](url)

cjue, 2018-09-19:
Hi,
as mentioned in issue #84, the voxel_side_length is given in meters.

The message "VoxelMap::insertPointCloud: You tried to insert points that lie outside the map dimensions!" is mostly informational, the points outside the specified voxel grid will silently be dropped.

The parallelBanding algorithm currently puts several constraints on the DistanceVoxelMap it operates on, most importantly "map_dimension.x == map_dimension.y".

The unit of map_dimension is voxels, with each voxel being a cube of size "voxel_side_length".

We plan to release a new alternative to DistanceKinectDemo directly subscribing to a ROS topic soon.

cjue, 2018-09-19:
I hope I did not close the issue prematurely - let me know if you think the parallelBanding output is erroneous.

closed 2018-09-19 by cjue

jlllweco, 2018-09-19:
Thanks a lot! I hope my code would help you for ROS topic.

jlllweco, 2018-09-21:
Sorry to bother you, I want to know if there anything wrong with my code to transport pointcould to get the metric_free_space.
 I want to get this value of every voxel but I can't get the right value.
 The values which are calculated out are always 
 73008 , 171986 or maybe the value of 2^n.
 The comment I wrote on the website is:
  
 Thanks a lot! However I also want to know what is metric_free_space mean and how is the metric_free_space transform from the real world to visualization map?
Can I get real time point cloud for mapping and calculate
metric_free_space = sqrtf(dv.squaredObstacleDistance(measurement_points[i])) * voxel_side_length;
for every voxel by replacing measurement_points[i]
by metric_free_space=dv.squaredObstacleDistance(Vector3i(map_dimensions.x,map_dimensions.y,map_dimensions.z));?
  
 Would you please help me figure out how to calculate the metric_free_space

 

 ------------------ 原始邮件 ------------------

cjue, 2018-11-09:
Dear Camilla,

your example looks correct:

`metric_free_space = sqrtf(dv.squaredObstacleDistance(measurement_points[i])) * voxel_side_length;
`
Regarding the PBA parameters: It is fine to use all default values for these parameters. They have a minor impact on performance and in some cases cause issues, if the map dimensions are not e.g. a clean multiple of 8.

Do you still find strange outputs when using parallelBanding3D?

tsijs, 2019-05-29:
@jlllweco I am considering doing something similar to what you described. Did you get this working nicely in the end? How was the runtime performance?

cjue, 2019-05-29:
@tsijs: The parallelBanding distance computation should work in a time proportional to the number of Voxels in the map. My guess would be about 10 times the duration of a device to device `cudaMemcpy`.

The new example `DistanceROSDemo` shows how to get the metric free space: https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/examples/DistanceROSDemo.cpp#L259

In a realistic application you would change the copy operation to copy all relevant voxels at once though,  instead of copying each voxel with a separate copy from device to host memory.

tsijs, 2019-07-12:
Sorry for the late response, I was busy with some other parts of a project that are fortunately working now:)
Which made me return to this. 
Let me maybe describe my setup and motivations for using gpu-voxels:
I want to have a dense distance field representation of the environment of my robot that I will use for trajectory optimization purposes. Since this optimization runs on the host, and should not have to wait for copying distance values from the device for every distance query it does, I want to copy the whole DistanceMap to the CPU every x ms (as fast as the hardware allows for it basically).
I am aware this will accumulate some delay in the system, but up to a certain level that is ok.

I have been able to run the DistanceROSDemo as you described which works nicely with my camera. However, I would like to be able to do the squaredObstacleDistance() operation on the whole map and get a copy of it to the host and I haven't found the right way to do that yet I think. I found the extract_distances() method in the API, which provides a free_space pointer. 
I found some related issues in #98, #84, and #65 where the first two seem to take the approach of the DistanceROSDemo and the last one recommends the use of copyCoordsToHost from the TemplateMap, though that might be only occupancy information?

In short, I am a bit lost in the API, if someone could give me some pointer in the right direction, I'd appreciate it.

cjue, 2019-07-31:
Hi @tsisj,

as you found one possibility is the extract_distances function. It will write the distance of each DistanceVoxel to its closest obstacle into a device array of freespace_t, with is an int8_t. When the distance is larger than 127, it is capped at that value. Distances smaller than radius are reported as 0.

The function is here:  
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/gpu_voxels/voxelmap/DistanceVoxelMap.hpp#L674
It's just a function object (located in DistanceVoxel.h) that is passed to a thrust::transform call that applies that function to every DistanceVoxel.

One application of the extract_distances function is in the DistanceConverter class:
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/gpu_voxels/helpers_highlevel/DistanceMapConverter.cu#L163

Please let me know if this is useful for your use case.