fzi-forschungszentrum-informatik/gpu-voxels #84: get index of each voxel and calculate EDT and then visualize the map

jlllweco, 2018-09-19:
I try to implement gpu-voxel to ROS workspace, and subscribe pointcloud topic "/zed/point_cloud/cloud_registered" data from zed-ros-wrapper ZEDmini camera.
But in the code,
pbaDistanceVoxmap->insertPointCloud(myKinectCloud, eBVM_OCCUPIED);
I try to take in real-time point-cloud to calculate pbaDistanceVoxmap.

The path of file is
gpu-voxels/packages/gpu_voxels/src/examples/DistanceKinectDemo.cpp
I want to get index of each voxel and calculate EDT and then visualize the map with our own programming using ROS/RVIZ.
Can it work with our ROS for visualization ? 

The code for visualization is 
` //visualization
            for (int x=0;x<map_dimensions.x;x++)
            {
                for (int y=0;y<map_dimensions.y;y++)
                {
                    for (int z=0;z<map_dimensions.z;z++)
                    {
                        // get id
                        int32_t id = voxelmap::getVoxelIndexSigned(map_dimensions,x,y,z);
                       
                        // get DistanceVoxel
                        DistanceVoxel dv; //get DistanceVoxel with closest obstacle information
                        
                        cudaMemcpy(&dv, (dvm_thrust_ptr+id).get(), sizeof(DistanceVoxel), cudaMemcpyDeviceToHost);
                        // get distance
                        //float metric_free_space = sqrtf(dv.squaredObstacleDistance(measurement_points[i])) * voxel_side_length;
                       
                        float metric_free_space=dv.squaredObstacleDistance(Vector3i(map_dimensions.x,map_dimensions.y,map_dimensions.z));
                        
                        // map distance to color
                        pcl::PointXYZRGBA clrP;
                        clrP.x = x;
                        clrP.y = y;
                        clrP.z = z;
                        clrP.r = 255;
                        clrP.g = 255;
                        clrP.b = 255;
                        clrP.a = metric_free_space;
                        pclOut->points.push_back (clrP);
                    }
                }
            }
            pcl_conversions::toPCL(ros::Time::now(), pclOut->header.stamp);
      
            pub_sp.publish (pclOut);
            pclOut->clear();` 

Would you please tell me how the index of `&dv, (dvm_thrust_ptr+id).get()`  calculate? 

what does the 256*256*256 and 0.01 mean in the real world? 
Thank you so much if you can help me.
[
[pcl_gpu_getid.tar.gz](https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/files/2395173/pcl_gpu_getid.tar.gz)
](url)

cjue, 2018-09-19:
Hi,

we have a function that computes the linear index corresponding to a given x/y/z coordinate:

https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/feb56b3faef263cf52b05d399fe270f98c273592/packages/gpu_voxels/src/gpu_voxels/voxelmap/kernels/VoxelMapOperations.h

int32_t id = voxelmap::getVoxelIndexSigned(map_dimensions,x,y,z);
// returns (z * dimensions.x * dimensions.y + y * dimensions.x + x)

With a voxel_side_length of 0.01 meters, the Limits of the voxel data-structure would be 2.56x2.56x2.56 meters.

jlllweco, 2018-09-19:
Thanks a lot! However I also want to know what is `metric_free_space` mean and how is the `metric_free_space` transform from the real world to visualization map?
Can I get real time point cloud for mapping and calculate 
`metric_free_space = sqrtf(dv.squaredObstacleDistance(measurement_points[i])) * voxel_side_length;`
for every voxel by replacing `measurement_points[i]` 
by `metric_free_space=dv.squaredObstacleDistance(Vector3i(map_dimensions.x,map_dimensions.y,map_dimensions.z));`?

cjue, 2019-07-31:
Hi,

sorry for missing your question.

The metric_free_space variable name comes from here:

https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/examples/DistanceROSDemo.cpp#L259

Computing the squared root of DistanceVoxel::squaredObstacleDistance returns a distance in voxel units. Multiplying this with the voxel side length in meters returns the obstacle distance in meters.

Side-note: The general assumption in GPU-Voxels is that pointcloud values are in meters and that a voxel_side_length of 0.01 corresponds to 1 cm.

The easiest way to get the distance of each voxel to its closest obstacle is currently DistanceVoxelMap::extract_distances:  https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/issues/83#issuecomment-516987092

closed 2019-07-31 by cjue