fzi-forschungszentrum-informatik/gpu-voxels #88: Distance map parameter meaning

husong998, 2018-10-15:
I followed the example code of ProbMap2DistMap.cpp to try to generate a distance map for my own use case. My probability map is simply a single dot, shown as below:
[image]

However, the generated probability map does not make sense to me:
[image]

We expect the distance map to be somewhat cubical or spherecal. The planar shape is really counter intuitive. Can someone explain to me why is it so?

The code snippet used to generate the distance map is as follow:
```
 // but we generat the distance map only from voxels with an occupancy higher than 0.55, so this section gets not converted
boost::shared_ptr<DistanceVoxelMap> dist_map(gvl->getMap("myDistanceMap")->as<DistanceVoxelMap>());
  dist_map->mergeOccupied(prob_map, Vector3ui(0), 0.55);
  dist_map->parallelBanding3D(m1, m2, m3, PBA_DEFAULT_M1_BLOCK_SIZE, PBA_DEFAULT_M2_BLOCK_SIZE, PBA_DEFAULT_M3_BLOCK_SIZE, 1);
```
What do the parameters of parallelBanding3D mean?
How is the occupancy value calculated?

Any help is appreciated.

cjue, 2018-11-10:
Hi,
thank you for your report!

Could you please show me some more of your example code?

Does the initial DistanceVoxelMap after mergeOccupied look as expected, if you don't run the parallelBanding3D step?

What is printed when you left-click on of the visible voxels?

Did you enter one of the "slice modes" by pressing "t" in the visualizer?

Regarding PBA parameters: m1,m2 and m3 are ways to increase the parallelism during the distance computation - for almost all 3D scenarios it's fine to have them at their default values of 1. Setting m3 to 2 or 4 can increase performance a bit.

cjue, 2018-11-12:
Hi again,

the ProbMap2DistMap example can be quite confusing, I will try to improve the comments in the file.

When you start the default example, you should see green voxels with an occupancy value of 0 (a probability of 0.5) and blue voxels with an occupancy of 125 (probability 0.992126). You can get this output on the visualizer commandline by left-clicking the voxels.

The example uses a threshold probability of 0.55 during the merge step, which will essentially cut out the "green" voxels.

Attached you can find the expected output for the Distance map. To get the correct rendering, right-click the visualizer and select "Render Moder > Distance Maps Rendermode > Multicolor gradient".

This should give you something like this: (the view is oriented towards the origin to show the gap on the backside of the cube.

[image]

You can also repeatedly use the "t" key to show just slices of the voxel data, like this:

[image]

Does your example behave as expected if you inspect it like this?

cjue, 2019-01-11:
Closing the issue beacause there was no reply. Please let me know if the problem still exists.

closed 2019-01-11 by cjue