fzi-forschungszentrum-informatik/gpu-voxels #92: Segmentation fault (core dumped) on gvl_ompl_planning

wendwosen, 2019-01-15:
Hello,

segmentation fault (core dumped) occur when running the executable gvl_ompl_planner. 
 cmake -DCMAKE_PREFIX_PATH="/home/oem/gpu-voxels/build/packages/icl_core;/home/oem/gpu-voxel/build/packages/gpu_voxels"
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
-- GVL include dirs are: /home/oem/gpu-voxels/packages/gpu_voxels/src
-- OMPL include dirs are: /opt/ros/kinetic/include
-- Configuring done
-- Generating done
-- Build files have been written to: /home/oem/gpu-voxels/gvl_ompl_planning
oem@lar-wm-desktop:~/gpu-voxels/gvl_ompl_planning$ make
[ 50%] Built target gvl_ompl_planner_helper
[100%] Built target gvl_ompl_planner
oem@lar-wm-desktop:~/gpu-voxels/gvl_ompl_planning$ 
oem@lar-wm-desktop:~/gpu-voxels/gvl_ompl_planning$ ./gvl_ompl_planner
Segmentation fault (core dumped)

I have no idea what this is.  But I am suspecting that it did not find the cuda package. missing from the list

I appreciate your kind help.

Squelsh, 2019-01-16:
Hi Wendwosen,

Did you have a look at 
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/issues/86
?

I assume incompatible C++11 defines.

wendwosen, 2019-01-21:
It worked after building without PCL support. Thank you.

cjue, 2019-01-23:
Glad we could help. Let us know if you have more feedback.

closed 2019-01-23 by cjue