fzi-forschungszentrum-informatik/gpu-voxels #93: Build error ‘pcl_isfinite’ was not declared in this scope

NorwegianPi, 2019-02-02:
Hello,

I have just started writing a master thesis on GPU collision detection using ROS. I came over GPU-Voxels and want to explore its possibilities, but i have some issues building the software.

I am running Ubuntu 16.04, CUDA v10.0.130  and ROS Kinetic (ros-kinetic-desktop-full).

Build stops at 63%, and i get the following message:

```
[ 63%] Building CXX object packages/gpu_voxels/src/gpu_voxels/octree/test/CMakeFiles/octree_provider.dir/Kinect.cpp.o
In file included from /home/ubuntugpu/git/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/test/Kinect.cpp:23:0:
/home/ubuntugpu/git/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/test/Kinect.h: In member function ‘void gpu_voxels::NTree::ONIWrapper::getConvertParameter(uint32_t&, uint32_t&, float&, float&, float&, float&)’:
/home/ubuntugpu/git/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/test/Kinect.h:118:44: error: ‘pcl_isfinite’ was not declared in this scope
if (pcl_isfinite (depth_focal_length_x_))
^
/home/ubuntugpu/git/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/test/Kinect.h:121:44: error: ‘pcl_isfinite’ was not declared in this scope
if (pcl_isfinite (depth_focal_length_y_))
^
/home/ubuntugpu/git/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/test/Kinect.h:124:47: error: ‘pcl_isfinite’ was not declared in this scope
if (pcl_isfinite (depth_principal_point_x_))
^
/home/ubuntugpu/git/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/test/Kinect.h:127:47: error: ‘pcl_isfinite’ was not declared in this scope
if (pcl_isfinite (depth_principal_point_y_))
^
packages/gpu_voxels/src/gpu_voxels/octree/test/CMakeFiles/octree_provider.dir/build.make:86: recipe for target 'packages/gpu_voxels/src/gpu_voxels/octree/test/CMakeFiles/octree_provider.dir/Kinect.cpp.o' failed
make[2]: *** [packages/gpu_voxels/src/gpu_voxels/octree/test/CMakeFiles/octree_provider.dir/Kinect.cpp.o] Error 1
CMakeFiles/Makefile2:4930: recipe for target 'packages/gpu_voxels/src/gpu_voxels/octree/test/CMakeFiles/octree_provider.dir/all' failed
make[1]: *** [packages/gpu_voxels/src/gpu_voxels/octree/test/CMakeFiles/octree_provider.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

It seems like it have some problems with pcl and pcl_isfinite. To the best of my knowledge I have installed all prerequisites. I would appreciate if someone could point me in the right direction to fix this issue.

Also attached cmake output if it is of any help. 

Cheers!!
Jarle Horneland

```
cmake ..
-- Detected version of GNU GCC: 54 (504)
-- icmaker dir is /home/ubuntugpu/git/gpu-voxels/icmaker
-- 
-- Library configuration for icmaker 1.0.0 =====================================
-- 
-- Checking for libraries not found in previous checks: 
-- 
CMake Warning (dev) at /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:65 (find_package):
  File /usr/share/cmake-3.5/Modules/FindBoost.cmake includes
  /home/ubuntugpu/git/gpu-voxels/icmaker/CMakeModules/FindThreads.cmake
  (found via CMAKE_MODULE_PATH) which shadows
  /usr/share/cmake-3.5/Modules/FindThreads.cmake.  This may cause errors
  later on .

  Policy CMP0017 is not set: Prefer files from the CMake module directory
  when including from there.  Run "cmake --help-policy CMP0017" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:1338 (find_dependency)
  icmaker/CMakeModules/FindBoost.cmake:51 (include)
  icmaker/CorePackages.cmake:37 (FIND_PACKAGE)
  icmaker/IcMaker.cmake:545 (INCLUDE)
  CMakeLists.txt:6 (INCLUDE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   date_time
[... 164 lines cut ...]
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- --------------------------------------------------------------------------
-- ------------------------ GPU Voxels configuration ------------------------
--   
-- [OK]      Building GPU-Voxels. Found CUDA.
-- [OK]      Building GPU-Voxels with visualization. GLEW, GLM, OpenGL and GLUT were found.
-- [OK]      Building GPU-Voxels with ROS connections. ROS was found.
-- [OK]      Building GPU-Voxels with URDF support. urdfdom, orocos and kdl_parser and ROS-kdl were found.
-- [OK]      Building GPU-Voxels with Kinect support. OpenNI was found.
-- [OK]      Building GPU-Voxels with Octomap support. Octomap found.
-- [OK]      Building GPU-Voxels with PCL interfaces. PCL found.
--   
-- -------------------- END of GPU Voxels configuration ---------------------
-- --------------------------------------------------------------------------
CMake Warning at packages/gpu_voxels/CMakeLists.txt:163 (MESSAGE):
  Building with THRUST_DEBUG option!

ICMAKER_CUDA_CPPDEFINES: -lineinfo --maxrregcount=63  -gencode=arch=compute_35,code=[compute_35] -DTHRUST_DEBUG -Xcompiler=-Wall
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntugpu/git/gpu-voxels/build

```

cjue, 2019-02-09:
Hi Jarle,

is it possible that you are using a custom PCL version, that is newer than the default from kinetic/16.04?

We use 16.04, kinetic and 10.0 in most of our workstations, without the pcl_isfinite error.

NorwegianPi, 2019-02-11:
Thanks,
Did a clean instal of ROS and PCL and build seems to work now.
Cheers!

closed 2019-02-11 by NorwegianPi