fzi-forschungszentrum-informatik/gpu-voxels #94: OpenGL Mathematics error: GLSL 4.10

arrfou90, 2019-02-07:
Hi!
I want to report this incompatibility error. I get a runtime error when running  `"gpu_voxels_visualizer"`

```
<****-**-** **:**:**.***>  Visualization(Warning) Visualizer::initializeVisualizer: Couldn't open the shared memory segment of Visualizer!

<****-**-** **:**:**.***> Shader(Error)::loadShaders: 0:1(10): error: GLSL 4.10 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES
```

My system:
**Ubuntu 18.04
CUDA 10.0 
Cuda_compute_cap:3.5
OpenGL version string: 3.0 Mesa 18.0.5**

Thanks,
Arrfou

arrfou90, 2019-02-08:
I resolved it by reinstalling Nvidia driver! apparently I had broken links.

closed 2019-02-08 by arrfou90

cjue, 2019-02-09:
Thank you for reporting your solution :)

CUN-bjy, 2022-01-13:
In my case, reinstallation was not enough..  Is anyone who has encountered it..?