fzi-forschungszentrum-informatik/gpu-voxels #95: The environment variable 'GPU_VOXELS_MODEL_PATH' could not be read

NorwegianPi, 2019-02-13:
Hi,

When I run the example code I get this error:
`Gpu_voxels_helpers(Error)::getGpuVoxelsPath: The environment variable 'GPU_VOXELS_MODEL_PATH' could not be read. Did you set it?`

The code still runs. What effect does this error have, and how should I set it?

When running './collisions' it can't find the path to 'robot4cmRes.pcd' unless I specify the full path in the code. Then everything seems to run ok with the path error, but it never detects any pointcloud collisions. Is this connected to the 'GPU_VOXELS_MODEL_PATH'? 

Hope it ok to publish this here.

Cheers, Jarle

wendwosen, 2019-02-13:
You have to set the environment variable 'GPU_VOXELS_MODEL_PATH' to the directory  'models' in the gpu voxels package. Replace  'path to gpu_voxels' to the directory where you kept gpu-voxels in the following command 

export GPU_VOXELS_MODEL_PATH='/path to gpu_voxels/gpu-voxels/packages/gpu_voxels/models/'

cjue, 2019-02-13:
@NorwegianPi: did this clear up the issue?

The environment variable is used in combination with relative paths for some examples and test.

NorwegianPi, 2019-02-13:
Yes, this solved the problem.
Thank you @wendwosen!

closed 2019-02-13 by NorwegianPi