fzi-forschungszentrum-informatik/gpu-voxels #97: Is Flood Fill algorithm is added into the library?

GPrathap, 2019-03-28:


cjue, 2019-04-02:
Hi, for the video you are referring to we implemented the final floodfill on the CPU. It was far from optimized.

The extract_distances and init_floodfill functions perform quite narrow optimizations, I will try to improve the documentation regarding this. 

The code for the video looked roughly like this:

```
voxelmap->parallelBanding3D();
voxelmap->extract_distances(dev_free_space_ptr, robot_min_radius);
voxelmap->init_floodfill(dev_free_space_ptr, dev_floodfill_distances_ptr, robot_min_radius);
cudaMemcpy dev_floodfill_distances_ptr to host_floodfill_distances
floodfill_cpu(host_floodfill_distances); //sequential implementation
```

I am closing this for now, but let me know if you have more questions.

closed 2019-04-02 by cjue

GPrathap, 2019-04-02:
Okay, thank you!. I will get back to you when something comes up.

GPrathap, 2019-04-02:
Can you give me a way to copy from GPU to CPU,

DistanceVoxel::extract_byte_distance::free_space_t* dev_free_space_ptr;
DistanceVoxel::init_floodfill_distance::manhattan_dist_t* dev_manhattan_distances_ptr;
pbaDistanceVoxmap->extract_distances(dev_free_space_ptr, 2.5);
pbaDistanceVoxmap->init_floodfill(dev_free_space_ptr, dev_manhattan_distances_ptr, 2.5);

// TODO cudaMemcpy dev_floodfill_distances_ptr to host_floodfill_distances 
This part is not clear?