fzi-forschungszentrum-informatik/gpu-voxels #98: How to get all extract_distances

GPrathap, 2019-04-08:
This problem is related to this: https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/issues/97

Can you give me a way to copy from GPU to CPU,

thrust::device_ptr<DistanceVoxel> dvm_thrust_ptr(pbaDistanceVoxmap->getDeviceDataPtr());
 
DistanceVoxel::extract_byte_distance::free_space_t* dev_free_space_ptr;
DistanceVoxel::init_floodfill_distance::manhattan_dist_t* dev_manhattan_distances__device_ptr;
DistanceVoxel::init_floodfill_distance::manhattan_dist_t* dev_manhattan_distances_host_ptr;
pbaDistanceVoxmap->extract_distances(dev_free_space_ptr, 2.5);
pbaDistanceVoxmap->init_floodfill(dev_free_space_ptr, dev_manhattan_distances__device_ptr, 2.5);

// TODO cudaMemcpy dev_floodfill_distances_ptr to host_floodfill_distances 
This part is not clear? 
I could try this way?

cudaMemcpy(&dev_manhattan_distances__device_ptr, dev_manhattan_distances_host_ptr, sizeof(DistanceVoxel::init_floodfill_distance::manhattan_dist_t), cudaMemcpyDeviceToHost);

How do you know the size of the vector to be copied?

cjue, 2019-04-08:
Hi Geesara,

please excuse my late reply. You just need to mutiply the bytes per voxel with the voxels per map. Also, it looks like you have an extra `&` before your first pointer.

```
size_t num_voxels = NXY*NXY*NZ; // dimensions of the distance voxelmap
using DistanceVoxel::init_floodfill_distance::manhattan_dist_t;
cudaMemcpy(dev_manhattan_distances__device_ptr, dev_manhattan_distances_host_ptr, sizeof(manhattan_dist_t) * num_voxels, cudaMemcpyDeviceToHost);
```

Let me know if you have more questions.

closed 2019-04-08 by cjue

cjue, 2019-07-31:
@GPrathap sorry for missing this earlier.

Hi, were you able to fix the problem? 

Did you omit from your example the cudaMalloc calls to allocate memory for device arrays and set  `dev_free_space_ptr` and `dev_manhattan_distances__device_ptr` to point to them?

GPrathap, 2019-08-01:
Yeap, I've gone with another approach which worked smoothly. Anyway thanks for the reply