fzi-forschungszentrum-informatik/gpu-voxels #101: 3D model generation and insertion problem

AlejoDiaz49, 2019-06-13:
Hello,

After a few months, I am retaking this part of my project. I left it aside because I was having a lot of problems with the 3D models of the robot. 

I've tried different approaches but everything failed. So I would like to know a few things:

- Which one is the unit for the 3D models? I have been using meters and everything is ok but when I use the `voxelize.sh` script to convert a file from `STL ` to `binvox`, in some models I always get 0 voxels

- In some cases when I visualize the `binvox` file using [viewvox](https://www.patrickmin.com/viewvox/) there are just a few voxels but is not like the shape of the real object. Is the scaling in `voxelize.sh` only for changing the size or is something different?

I tried with both `XYZ` and `PCD` files from the `STL ` model, but I have some problems placing the coordinates system and sometimes the model doesn't even appear in the visualizer.

How do you convert and adapt the models to work perfectly in your implementations? I would appreciate the help because this is the only thing that until now I'm not able to make work in this library.

I've been using Inventor to get the STL file with the correct coordinates system and then either directly to binvox using the script or CloudCompare to get `XYZ` or `PCD`. Just want to mention that I'm using the DH parameters to insert the robots

Thanks in advanced
Alejandro

cjue, 2019-06-14:
Hi Alejandro,

the STL file does not specify any scale in itself: https://en.wikipedia.org/wiki/STL_(file_format)

Tools like Blender allow you to set a scale before exporting to STL: https://blender.stackexchange.com/questions/7503/scale-settings-for-exporting-to-stl-for-3d-printing

I am not an expert regarding binvox and voxelize.sh - my guess would be, that the problematic meshes were exported in meters and you need to use the scaling parameter of voxelize.sh to get a useful result. Please let me know if your problem with voxelization persists.

AlejoDiaz49, 2019-07-05:
Hi Christian,

At the end it was a problem with the program, it was messing the STL files, changing the scale and coordinates origin.

At the end, I used 3DS Max from Autodesk and it works directly without problems and then running the voxelize.sh script in mm

Thanks and sorry for opening an issue for this
Alejandro

closed 2019-07-05 by AlejoDiaz49

cjue, 2019-07-05:
No problem, 

please let me know when you find other issues :)