fzi-forschungszentrum-informatik/gpu-voxels #103: Build error related to gpu_voxels_voxelmap

mehmetkillioglu, 2019-09-26:
Hi,

I am trying to build gpu-voxels on Ubuntu 16.04. Cuda 10.1 with Ros kinetic. CMake and make output are below. Thanks in advance. 

Cmake Output:
```
-- Detected version of GNU GCC: 54 (504)
-- icmaker dir is /home/mehmet/code/gpu-voxels/icmaker
-- 
-- Library configuration for icmaker 1.0.0 =====================================
-- 
-- Checking for libraries not found in previous checks: 
-- 
CMake Warning (dev) at /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:65 (find_package):
  File /usr/share/cmake-3.5/Modules/FindBoost.cmake includes
  /home/mehmet/code/gpu-voxels/icmaker/CMakeModules/FindThreads.cmake (found
  via CMAKE_MODULE_PATH) which shadows
  /usr/share/cmake-3.5/Modules/FindThreads.cmake.  This may cause errors
  later on .

  Policy CMP0017 is not set: Prefer files from the CMake module directory
  when including from there.  Run "cmake --help-policy CMP0017" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:1338 (find_dependency)
  icmaker/CMakeModules/FindBoost.cmake:51 (include)
  icmaker/CorePackages.cmake:37 (FIND_PACKAGE)
  icmaker/IcMaker.cmake:545 (INCLUDE)
  CMakeLists.txt:6 (INCLUDE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   date_time
--   filesystem
--   graph
--   iostreams
--   system
--   regex
--   signals
--   unit_test_framework
--   thread
--   program_options
--   serialization
--   chrono
--   atomic
-- Found GLUT: ... yes
-- Could not find GSL
-- Iconv: Could not find required library iconv.
-- LibArchive: Could not find required header file archive.h.
-- Could NOT find libwebsockets
-- Ncomrx: Could not find required header file NComRxC.h.
-- Newmat: Could not find required header file newmat/newmat.h.
-- Could NOT find SCIP
-- Could NOT find speechd
-- Could NOT find Tcmalloc
-- Could NOT find XSD
-- Spacenav: Could not find required header file spnav.h.
-- === Searching Perception packages ===
-- Found OpenCV: /opt/ros/kinetic (found version "3.3.1") 
-- Found OpenCV: ... yes
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.
[... 367 lines cut ...]

/home/mehmet/code/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/voxelmap/DistanceVoxelMap.hpp(61): error: not a class or struct name

/home/mehmet/code/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/voxelmap/DistanceVoxelMap.hpp(61): error: class or struct definition is missing

/home/mehmet/code/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/voxelmap/DistanceVoxelMap.hpp(70): warning: parsing restarts here after previous syntax error

3 errors detected in the compilation of "/tmp/tmpxft_00005424_00000000-6_VoxelMap.cpp1.ii".
CMake Error at gpu_voxels_voxelmap_CUDA_TARGET_generated_VoxelMap.cu.o.cmake:266 (message):
  Error generating file
  /home/mehmet/code/gpu-voxels/build/packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap_CUDA_TARGET.dir//./gpu_voxels_voxelmap_CUDA_TARGET_generated_VoxelMap.cu.o

packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap.dir/build.make:3790: recipe for target 'packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap_CUDA_TARGET.dir/gpu_voxels_voxelmap_CUDA_TARGET_generated_VoxelMap.cu.o' failed
make[2]: *** [packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap_CUDA_TARGET.dir/gpu_voxels_voxelmap_CUDA_TARGET_generated_VoxelMap.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:4661: recipe for target 'packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap.dir/all' failed
make[1]: *** [packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 58%] Building CXX object packages/gpu_voxels/src/gpu_voxels/voxelmap/CMakeFiles/gpu_voxels_voxelmap_tests.dir/Dummy.cpp.o
[ 58%] Linking CXX shared library ../../../../../lib/libgpu_voxels_voxelmap_tests.so
[ 58%] Built target gpu_voxels_voxelmap_tests
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

```

cjue, 2019-09-26:
Hi Mehmet,

thanks for the report. I will look into why this breaks with Cuda 10.1.

As a quick-fix, you can add this to src/gpu_voxels/voxelmap/DistanceVoxelMap.hpp:
```
#include <thrust/device_malloc_allocator.h>
```

mehmetkillioglu, 2019-09-26:
Thanks for the quick response. I was just about to make this comment: 

Turns out 
```
#include <thrust/device_malloc_allocator.h>
```
was missing from the file 
```
./gpu-voxels/packages/gpu_voxels/src/gpu_voxels/voxelmap/DistanceVoxelMap.hpp
```

Adding the include line solved the issue.

cjue, 2019-09-26:
Thanks for the update. Will be fixed in the upcoming release, which is currently going through testing.

commit 4dcb7223 (2019-10-28): Merge pull request #104 from UM-ARM-Lab/Cuda10.1_Patch

Fix: Add include for compiling with cuda10.1. Also fixes #103
closed 2019-09-26 by cjue