fzi-forschungszentrum-informatik/gpu-voxels #106: Raycasting  leave free space As Occupied

wendwosen, 2020-01-29:
Hi @cjue 

I wanted to ask about  insertPointcloudWithFreeSpaceCalculation method.

After insertion new pointclouds using this method, there are voxels left in the map that are actually not occupied. 
This the call to the function in my code:
 gvl->getMap("myEnvironmentMap")->as<NTree::GvlNTreeDet>()->insertPointCloudWithFreespaceCalculation(mycloud, mycamera_offsets, 8, 8);

Can you suggest a proper way to select free_space_resolution and occupied_space_resolution parameters. I suspect that is the problem. 

 my choice of 8mm for both is exactly equals to the size of voxels during initialization. 

best regards

cjue, 2020-01-29:
Hi @wendwosen,

I have an internal fix for `insertPointcloudWithFreeSpaceCalculation` that I will try to bring to Github as soon as possible, that could also solve your problem.

One question: What GPU model are you using for this?

wendwosen, 2020-01-29:
Thank you @cjue  for the reply.

The gpu in my computer is GP102 [GeForce GTX 1080 Ti]

insertSensorData is also able to perform raycasting but it doesnt work at all for me.

Can you recommend me a method to insert pointcloud with raycasting that works for you too. 

I look forward to the fix.

Redox15, 2020-04-13:
Any news on that?
I'm using the ROS example and I can't find a way to insert PointClouds with raycasting.