fzi-forschungszentrum-informatik/gpu-voxels #111: Incorrect occupancy when varying voxel side length

mfinean, 2020-09-17:
The following produces 16777216 occupied voxels = 256^3 as expected.
```
    Vector3ui map_dimensions(256, 256, 256);
    float voxel_side_length = 1.0f;

    Vector3f env_min(0,0,0);
    Vector3f env_max(255, 255, 255);    
    std::vector<Vector3f> envPoints = gpu_voxels::geometry_generation::createBoxOfPoints(env_min, env_max, voxel_side_length);
    countingVoxelList->insertPointCloud(envPoints, eBVM_OCCUPIED);
```

If we instead use a voxel side length of 0.01, we get 16387064 = 254^3 voxels
```
    Vector3ui map_dimensions(256, 256, 256);
    float voxel_side_length = 0.01f;

    Vector3f env_min(0,0,0);
    Vector3f env_max(2.55, 2.55, 2.55);    
    std::vector<Vector3f> envPoints = gpu_voxels::geometry_generation::createBoxOfPoints(env_min, env_max, voxel_side_length);
    countingVoxelList->insertPointCloud(envPoints, eBVM_OCCUPIED);
```

Any ideas why this is?

Squelsh, 2020-09-17:
If I remember right, this pretty simple helper function just starts at the given coordinates, and iteratively adds the voxel_side_length until it hits the second coordinates. What you see is the fringe case where it hits the upper boundary exactly. Due to floating point precision instability you may see the shown results.
My advice would be to start of with coordiantes half a voxel_side_length off if your volume is a multiple of the side-length:
Vector3f env_min(0.5*voxel_side_length,0.5*voxel_side_length,0.5*voxel_side_length);

So each probing points falls into the center of a voxel and the situation is clear.

cjue, 2020-09-17:
Yes exactly, the geometry_generation functions that output points with floating point coordinates will be converted to voxels during the `insertPointCloud `function. The workaround described by @Squelsh will achieve exactly what you want: add half the `voxel_side_length `to both the min and max boundary coordinates.

https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/gpu_voxels/helpers/GeometryGeneration.cpp#L90

We could address this by changing the comparison with the boundary to a more floating point friendly version:
`fabs(x_dim - max.x) < tolerance` instead of `x_dim <= max.x`

Although: this only explains why there would be 255^3 voxels instead of 256^3. i will look into it more closely.

There is also a second version of the geometry helper function that outputs integer voxel coordinates than can be fed to the voxellist `insertCoordinateList` functions.

cjue, 2020-09-18:
In the case with voxel_side_length 0.01f many points fall into the same voxel. because the floating point representation is smaller than "0.01".

Adding half the voxel_side_length to min and max fixes this:

```
  {
    float voxel_side_length = 1.0f;
    gpu_voxels::voxellist::CountingVoxelList list1(Vector3ui(256, 256, 256), voxel_side_length, MT_COUNTING_VOXELLIST);
    float h = voxel_side_length/2;
    Vector3f env_min(0+h, 0+h, 0+h);
    Vector3f env_max(255+h, 255+h, 255+h);
    std::vector<Vector3f> envPoints = gpu_voxels::geometry_generation::createBoxOfPoints(env_min, env_max, voxel_side_length);
    list1.insertPointCloud(envPoints, eBVM_OCCUPIED);
    std::cout << "Number of voxels: " << list1.getDimensions().x << std::endl;
    list1.remove_underpopulated(1);
    std::cout << "Number of voxels: " << list1.getDimensions().x << std::endl;
    list1.remove_underpopulated(2);
    std::cout << "Number of voxels: " << list1.getDimensions().x << std::endl;
    list1.remove_underpopulated(3);
    std::cout << "Number of voxels: " << list1.getDimensions().x << std::endl;
    list1.remove_underpopulated(4);
    std::cout << "Number of voxels: " << list1.getDimensions().x << std::endl;
  }
  {
    float voxel_side_length = 0.01f;
    gpu_voxels::voxellist::CountingVoxelList list1(Vector3ui(256, 256, 256), voxel_side_length, MT_COUNTING_VOXELLIST);
    float h = voxel_side_length/2;
    Vector3f env_min(0+h, 0+h, 0+h);
    Vector3f env_max(2.55+h, 2.55+h, 2.55+h);
    std::vector<Vector3f> envPoints = gpu_voxels::geometry_generation::createBoxOfPoints(env_min, env_max, voxel_side_length);
    list1.insertPointCloud(envPoints, eBVM_OCCUPIED);
    std::cout << "Number of voxels: " << list1.getDimensions().x << std::endl;
    list1.remove_underpopulated(1);
    std::cout << "Number of voxels: " << list1.getDimensions().x << std::endl;
    list1.remove_underpopulated(2);
    std::cout << "Number of voxels: " << list1.getDimensions().x << std::endl;
    list1.remove_underpopulated(3);
    std::cout << "Number of voxels: " << list1.getDimensions().x << std::endl;
    list1.remove_underpopulated(4);
    std::cout << "Number of voxels: " << list1.getDimensions().x << std::endl;
  }
```

There is also a second version of createBoxOfPoints with an additional argument that outputs integer voxel coordinates, I should improve the documentation to make the differences and the pitfalls with the floating point version clearer.

Closing this now, let me know if you have further questions.

closed 2020-09-18 by cjue