fzi-forschungszentrum-informatik/gpu-voxels #112: CudaLog(Error)::cuCheckForError: too many resources requested for launch(7)

diybl, 2020-09-29:
i run the heightmap example on jetson nano which have only 128 cuda cores.
it shows the error

<2020-09-29 15:40:22.879> CudaLog(Error)::cuCheckForError: too many resources requested for launch(7) in /root/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp on line 557.
<2020-09-29 15:40:22.971> CudaLog(Error)::cuHandleError: unspecified launch failure in /root/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp on line 616.
<2020-09-29 15:40:22.972> CudaLog(Error)::cuCheckForError: unspecified launch failure(4) in /root/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp on line 557.
<2020-09-29 15:40:22.972> CudaLog(Error)::cuHandleError: unspecified launch failure in /root/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp on line 558.

cjue, 2020-09-29:
Hi @diybl,

did you have a look at the jetson-related line in CMakeLists.txt?
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/CMakeLists.txt#L156

This need to be lowered to 31 for Jetson TX1 and TX2 and the Jetson Nano has the same limitations.

This is probably the same error:
https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/issues/78#issuecomment-408592663

diybl, 2020-09-29:
now it works
thanks for your help~~

cjue, 2020-09-29:
Happy to help. Let me know if you find further issues on the Nano.

closed 2020-09-29 by cjue