fzi-forschungszentrum-informatik/gpu-voxels #114: libgpu_voxels_visualization_core.so, not found

diybl, 2020-10-04:
i compile my program with gpu-voxels lib,
link error is this:

[ 33%] Building CXX object cispex/CMakeFiles/cispex.dir/gvl_linkage_test_lib.cpp.o
[ 66%] Building CXX object cispex/CMakeFiles/cispex.dir/cispex.cpp.o
[100%] Linking CXX executable ../bin/cispex
/usr/bin/ld: warning: libvtkglew-7.1.so.1, needed by /root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so, not found (try using -rpath or -rpath-link)
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewBindBuffer'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `glewExperimental'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewBufferData'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewUniform3fv'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewGetShaderiv'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewDisableVertexAttribArray'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewGenVertexArrays'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewDrawArraysInstanced'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewVertexAttribDivisor'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewShaderSource'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewDeleteShader'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewDrawElementsInstanced'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewVertexAttribPointer'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewBindVertexArray'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewUseProgram'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewGetUniformLocation'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewUniform4f'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewGetProgramiv'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewGetShaderInfoLog'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewGenBuffers'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `glewGetErrorString'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewUniformMatrix4fv'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewCreateProgram'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewDeleteBuffers'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewCompileShader'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewPrimitiveRestartIndex'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewUniform1i'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewAttachShader'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewGetProgramInfoLog'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewLinkProgram'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewUniform4fv'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewEnableVertexAttribArray'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `__glewCreateShader'
/root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so: undefined reference to `glewInit'
collect2: error: ld returned 1 exit status
cispex/CMakeFiles/cispex.dir/build.make:123: recipe for target 'bin/cispex' failed
make[2]: *** [bin/cispex] Error 1
CMakeFiles/Makefile2:85: recipe for target 'cispex/CMakeFiles/cispex.dir/all' failed
make[1]: *** [cispex/CMakeFiles/cispex.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

cjue, 2020-10-04:
Hi,

I have not yet seen that error. What operating system version are you using?

diybl, 2020-10-05:
> Hi,
> 
> I have not yet seen that error. What operating system version are you using?

jetson nano ubuntu

this is my cmaklist

cmake_minimum_required (VERSION 3.8)

list(APPEND CMAKE_PREFIX_PATH "/root/gpu-voxels/export")

include_directories (
    PUBLIC ${CMAKE_CURRENT_SOURCE_DIR}
    PUBLIC ${gpu_voxels_INCLUDE_DIRS}
    PUBLIC ${orocos_kdl_INCLUDE_DIRS} # this should be removed in upcoming versions.
    PUBLIC ${CUDA_INCLUDE_DIRS}
)

aux_source_directory(. DIR_SRCS)

add_executable(cispex ${DIR_SRCS})

target_link_libraries (cispex
    LINK_PUBLIC ${Boost_SYSTEM_LIBRARY}
    LINK_PUBLIC ${icl_core_LIBRARIES}
    LINK_PUBLIC ${gpu_voxels_LIBRARIES}
    LINK_PUBLIC ${CUDA_LIBRARIES}
)

cjue, 2020-10-05:
The gpu_voxels_visualizer is not supported on the Jetson platform.

This should not interfering with linking to gpu_voxels through, so I don't understand how your CMakeLists.txt can lead to the error output you provided:

> `/usr/bin/ld: warning: libvtkglew-7.1.so.1, needed by /root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so, not found`

`libgpu_voxels_visualization_core` is not part of the `gpe_voxels` target.

diybl, 2020-10-05:
but i complie it only。not run

diybl, 2020-10-05:
> The gpu_voxels_visualizer is not supported on the Jetson platform.
> 
> This should not interfering with linking to gpu_voxels through, so I don't understand how your CMakeLists.txt can lead to the error output you provided:
> 
> > `/usr/bin/ld: warning: libvtkglew-7.1.so.1, needed by /root/gpu-voxels/export/lib/libgpu_voxels_visualization_core.so, not found`
> 
> `libgpu_voxels_visualization_core` is not part of the `gpu_voxels` target.

it seems that this problem
https://stackoverflow.com/questions/31438916/cmake-cannot-find-library-using-link-directories

and i found  the /share/icl_core/icl_core-config.cmake
it use the link_directories 

my cmake version is 3.10.2

diybl, 2020-10-06:
now i solved it
that is the vtk7.1.1 problem witch i compiled it on my machine.

cmakelist add this
SET(VTK_DIR "/root/VTK-7.1.1/build")
FIND_PACKAGE(VTK REQUIRED)

add the lib

LINK_DIRECTORIES(${vtkalglib_RUNTIME_LIBRARY_DIRS})

build success～～

closed 2020-10-06 by diybl