fzi-forschungszentrum-informatik/gpu-voxels #115: how do we vizualize anything from d435i camera using ros topic

AndreV84, 2020-10-13:
we got the vizualizer, but we couldn't vizualize in it anything how should be test it to see it draws anything?

AndreV84, 2020-10-13:
[log.txt](https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/files/5373152/log.txt)

AndreV84, 2020-10-13:
upd we could vizualize something, but the question is how to get d435i intel camera topics from ROS vizualized

cjue, 2020-10-13:
Hi, the visualizer itself can only visualize the GPU data structures of other GPU-Voxels processes.

So to actually subscribe to the camera point cloud topic, you also need to run another program like `distance_ros_demo`.

Related links:
* blog post introducing distance_ros_demo and the example rosbags:
  https://www.gpu-voxels.org/release-1-3-0-merging-and-morphing/
* Comments with details for this example program: 
  https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/examples/DistanceROSDemo.cpp#L23

Please let me know if you have further questions.

AndreV84, 2020-10-13:
when we used to run both the distance_ros_demo with gpu_vizualizer it would show the window from the ros_distance, but it would just draw three points[ filled circles having various colours there.
How to get it to draw full image from d435 camera vi athe ros_distance? are there particular steps to do so?

AndreV84, 2020-10-13:
it wouldn't execute sudo make install
what are the particular reasons behind it?

cjue, 2020-10-14:
> it wouldn't execute sudo make install
> what are the particular reasons behind it?

To be honest we don't use `sudo make install` in our workflow. `make install` should place everything in the `export/` folder.
We add that export folder to the search path of other projects linking in GPU-Voxels as a library.

cjue, 2020-10-14:
> when we used to run both the distance_ros_demo with gpu_vizualizer it would show the window from the ros_distance, but it would just draw three points[ filled circles having various colours there. [...]

It sounds like you need to specify the correct topic to subscribe to for your camera. You can use the `-t /topic/with/the/points` command line argument to specify it, as specified here:

https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/examples/DistanceROSDemo.cpp#L114

AndreV84, 2020-10-14:
@cjue Thank you for following up. We are trying to use gpu_voxels both on x86_64 and Nvidia Jetson AGX.Do you know if as for today the gpu_vizualization binary works vizualizing outputs from any binaries, if executed on Jetson arm device?
Thank you very much

AndreV84, 2020-10-14:
when I build it on xavier that has 72SM architecture while executing
make install
it appears to show
```
------------------ END of GPU Voxels configuration ---------------------
-- --------------------------------------------------------------------------
CMake Warning at packages/gpu_voxels/CMakeLists.txt:163 (MESSAGE):
  Building with THRUST_DEBUG option!

ICMAKER_CUDA_CPPDEFINES: -lineinfo --maxrregcount=63  -gencode=arch=compute_35,code=[compute_35] -DTHRUST_DEBUG -Xcompiler=-Wall

```
does it look ok or I should change somewhere the value?

cjue, 2020-10-15:
> @cjue Thank you for following up. We are trying to use gpu_voxels both on x86_64 and Nvidia Jetson AGX.Do you know if as for today the gpu_vizualization binary works vizualizing outputs from any binaries, if executed on Jetson arm device?
> Thank you very much

The Jetson devices are based on an ARM architecture and lack support for the type of shared-memory between processes used by the visualizer. Therefore the gpu_voxels_visualizer will not work on Jetson devices for the foreseeable future.

The alternative is to publish data, e.g. for visualization in RViz, as mentioned in #65.

cjue, 2020-10-15:
> ICMAKER_CUDA_CPPDEFINES: -lineinfo --maxrregcount=63  -gencode=arch=compute_35,code=[compute_35] -DTHRUST_DEBUG -Xcompiler=-Wall
>
> does it look ok or I should change somewhere the value?

This will work fine. You could change the ICMAKER_CUDA_COMPUTE_VERSION to `72` in packages/gpu_voxels/CMakeLists.txt to avoid any additional compilation during the first program launch after compilation, but it should all work the same.

The minimum architecture level required by GPU-Voxels is 3.5.

AndreV84, 2020-10-19:
Thank you for following up!
Are there any pointers if we could get aa .ply file or .ply file converted in binvox e.g. as with https://github.com/Forceflow/cuda_voxelizer to be vizualized somehow with the vizualizer or it will require to write a launching binary from scratch? Thank you very much!

AndreV84, 2020-10-23:
@cjue could we in a simple way play binvox file somehow?
Thanks

cjue, 2020-10-24:
Here is an example for inserting a bin ox file:

  gvl->insertPointCloudFromFile("myObjectVoxellist", "/schunk_svh/f20.binvox", true,

AndreV84, 2020-10-24:
@cjue may I know if there is any particular source example file to edit, please? so that that it will be possible to add the line above, then rebuild?
Thank you very much!

amferreiraBEV, 2020-10-28:
Hi @AndreV84,
Me and my team are using a Realsense d435i and ProbVoxelMap to vizualize the data from it.
We are also working with Jetson (Nano and Xavier).
If using GPU-Voxels and d435i together is what you want, contact me by e-mail!

AndreV84, 2020-10-28:
@amferreiraBEV 
Thank you for your response.
I will try to google your email as the github doesn't seem to show it.
It would be great to get Jetson with d435i to get the camera outputs visualized with GPU-Voxels
Thank you very much

amferreiraBEV, 2020-10-28:
> @amferreiraBEV
> Thank you for your response.
> I will try to google your email as the github doesn't seem to show it.
> It would be great to get Jetson with d435i to get the camera outputs visualized with GPU-Voxels
> Thank you very much

Sorry, I thought that my e-mail were public (as it is now).

We're working on a way to visualize ProbVoxelMaps on Rviz since the gpu_voxels_visualizer don't work on Jetson as mention by @cjue.

AndreV84, 2020-10-29:
we can install on Jetson with
```
wget https://codeload.github.com/amferreiraBEV/gpu-voxels/zip/BEV_version
mv BEV_version BEV_version.zip
unzip BEV_version.zip
cd gpu-voxels-BEV_version/
mkdir build
cd build
source /opt/ros/melodic/setup.bash
cmake ..
make -j8
make install
```
However, the next steps should probably be to run roscore,
play rosbag/ or publish topic from d435i device, then just stat the rviz with rviz command?
Am I missing some steps?

AndreV84, 2020-11-04:
what if we run d435 ros topic on Jetson
but vizualize it over tcp on x86_64?
will the ros_disance example need heavy modification for it?

amferreiraBEV, 2020-11-04:
> what if we run d435 ros topic on Jetson
> but vizualize it over tcp on x86_64?
> will the ros_disance example need heavy modification for it?

Then the GPU-Voxels will run on x86_64 and not on Jetson!
But you will need a good network.

AndreV84, 2020-11-04:
we do not know how to complete the puzzle on jetson yet
although directoins were guven we need more exact understanding probably
maybe x86_64 vizualization makes sense e.g. if the camera is on jetson robot but the vizualizer runs on x86 PC,
Again the complication is we do not know exactly which ros disance code to modify if any although it might be implemented probably solely via ros mapping of topics

amferreiraBEV, 2020-11-04:
> we do not know how to complete the puzzle on jetson yet
> although directoins were guven we need more exact understanding probably
> maybe x86_64 vizualization makes sense e.g. if the camera is on [...]

I told you what you need to add to the code and I sent you my main loop.

[image]
[image]

Can't you see the differences?

AndreV84, 2020-11-04:
thank you for sharing the code
Howverm could you extend which exactly lines [ line numbers] in which exactly file needs to be substituted? added? edited?]
I still see that jetson wil not pass throghh the shared memory
it might send to rviz but we using separate topic publishing for doing it

amferreiraBEV, 2020-11-05:
> thank you for sharing the code
> Howverm could you extend which exactly lines [ line numbers] in which exactly file needs to be substituted? added? edited?]
> I still see that jetson wil not pass throghh the shared memory
> it might send to rviz but we using separate topic publishing for doing it

As I told you by e-mail, the lines you should add are:
`erodeTempVoxmap2->publishPointcloud(&pubPointCloud, paramOccupancy);`
`pubPCD.publish(pubPointCloud);`

AndreV84, 2020-11-05:
which are the numbers of the lines to that to add these?
shall they be added [appendded] or added via substitution?
in which exactly file they need to be put? from which reository? the original or the BEV fork [ modified]?
Thank you very much

amferreiraBEV, 2020-11-05:
> which are the numbers of the lines to that to add these?
> shall they be added [appendded] or added via substitution?
> in which exactly file they need to be put? from which reository? the original or the BEV fork [ modified]?
> Thank you very much

Look at those two images and figure it out for your self!

AndreV84, 2020-11-08:
soes it use cuda on Jetson?
https://docs.nvidia.com/cuda/cuda-for-tegra-appnote/index.html#memory-selection
with rviz?
Thanks

AndreV84, 2020-11-16:
we were able to vizualize in RVIZ d435 camera stream using the solution proposed by @amferreiraBEV 
Do you guys gonna to add further on the issue or it could be closed as resolved?
Thank you very much.

AndreV84, 2020-11-22:
is it likely that we could run the app with cuda 11.1 ?
Thanks

AndreV84, 2020-11-22:
```
/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/PointCloud.cu:22:10: fatal error: thrust/system/cuda/detail/cub/device/device_radix_sort.cuh: No such file or directory
 #include <thrust/system/cuda/detail/cub/device/device_radix_sort.cuh>
```
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

I can see the file is there:

/usr/local/cuda-11.1/targets/x86_64-linux/include/cub/device/device_radix_sort.cuh

AndreV84, 2020-11-22:
tried manually symlinking paths:
```
gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/PointCloud.cu(25): error: namespace "thrust::cuda_cub" has no member "cub"

nvcc warning : The 'compute_35', 'compute_37', 'compute_50', 'sm_35', 'sm_37' and 'sm_50' architectures are deprecated, and may be removed in a future release (Use -Wno-deprecated-gpu-targets to suppress warning).
/usr/local/cuda-11.1/include/cub/device/dispatch/dispatch_scan.cuh(50): error: this declaration has no storage class or type specifier

/usr/local/cuda-11.1/include/cub/device/dispatch/dispatch_scan.cuh(53): error: expected a ";"

/usr/local/cuda-11.1/include/cub/device/dispatch/dispatch_scan.cuh(166): warning: parsing restarts here after previous syntax error

/usr/local/cuda-11.1/include/cub/device/dispatch/dispatch_scan.cuh(167): error: expected a declaration

/usr/local/cuda-11.1/include/cub/device/dispatch/dispatch_scan.cuh(170): error: ChainedPolicy is not a template

/usr/local/cuda-11.1/include/cub/device/dispatch/dispatch_scan.cuh(170): error: identifier "Policy130" is undefined

/usr/local/cuda-11.1/include/cub/device/dispatch/dispatch_scan.cuh(170): error: not a class or struct name

/usr/local/cuda-11.1/include/cub/device/dispatch/dispatch_scan.cuh(173): error: AgentScanPolicy is not a template