fzi-forschungszentrum-informatik/gpu-voxels #119: Fix Issues with Ubuntu 20.04

AlejoDiaz49, 2020-12-28:
Hello everyone,

I am migrating the Framework of my team to Ubuntu 20, there are not really many issues with GPUVoxels, but quite a lot warnings. This my current environment.

### Basic Dependencies

| Library | Previous Version | New Version |
|:---------:|:-----------------------:|:-----------------------------:|
| g++ | 4:7.4.0-1ubuntu2.3 | 4:9.3.0-1ubuntu2 |
| gcc| 4:7.4.0-1ubuntu2.3 | 4:9.3.0-1ubuntu2 |
| make | 4.1-9.1ubuntu1 | 4.2.1-1.2 |
| cmake | 3.10.2-1ubuntu2.18.04.1 | 3.16.3-1ubuntu1 |

### Dependencies

| Library | Previous Version | New Version |
|:---------:|:-----------------------:|:-----------------------------:|
| CUDA | 10.0 | 11.0 |
| Boost | 1.65.1.0ubuntu1 | 1.71.0.0ubuntu2 |
| Eigen | 3.3.7 | 3.3.9 |
| FLANN | 1.9.1+dfsg-2 | 1.9.1+dfsg-9build1 |
| TinyXML | 2.6.2-4 | 2.6.2-4build1 |
| PCL | 1.9.1 | 1.9.1|
| VTK | 7.1.1+dfsg1-2 | 7.1.1+dfsg2-2ubuntu1 |
| FreeGLUT | 2.8.1-3 | 2.8.1-3 |
| GLM | 0.9.9~a2-2 | 0.9.9.7+ds-1 |
| GLEW | 2.0.0-5| 2.1.0-4 |

Although PCL has newer versions, I have to make many changes in my own code to make it work, I will change it whenever I can this year. So for the moment I will stick with 1.9.1 but the other things I will change them.

Also for CUDA the one that come with [nvidia-cuda-toolkit](https://packages.ubuntu.com/focal/nvidia-cuda-toolkit) is 10.1 but this one only work with gcc 8 or less, and I don't want to have different version of gcc since the default on Ubuntu 20 is gcc 9 and I already had some issues trying to compile everything

In any case I will try to build a Docker image with this environment and put here the file, so that is easier to reproduce.

Regarding the issues, I only got 2 errors:

## CUDA 11
The error is
````bash
/home/alejandro/Libraries/gpuvoxels/packages/gpu_voxels/src/gpu_voxels/octree/PointCloud.cu:22:10: fatal error: thrust/system/cuda/detail/cub/device/device_radix_sort.cuh: No such file or directory
   22 | #include <thrust/system/cuda/detail/cub/device/device_radix_sort.cuh>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
````
I tested the merge request #117 and It seems to compile and pass the tests. Will it be merge?
## Boost 1.71
The error is
````bash
/home/alejandro/gpu-voxels/packages/gpu_voxels/src/gpu_visualization/Visualizer.cu(1687): error: no instance of constructor "boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration [with base_duration=boost::posix_time::time_duration, frac_of_second=1000L]" matches the argument list
            argument types are: (float)

1 error detected in the compilation of "/home/alejandro/gpu-voxels/packages/gpu_voxels/src/gpu_visualization/Visualizer.cu".
CMake Error at gpu_voxels_visualization_core_CUDA_TARGET_generated_Visualizer.cu.o.RelWithDebInfo.cmake:280 (message):
  Error generating file
  /home/alejandro/gpu-voxels/build/packages/gpu_voxels/src/gpu_visualization/CMakeFiles/gpu_voxels_visualization_core_CUDA_TARGET.dir//./gpu_voxels_visualization_core_CUDA_TARGET_generated_Visualizer.cu.o
````
The problem i think is that this only accepts _int_ values now, to test it I put a cast from _float_ to _int_ and it compiled, although is probably not the correct solution. I am not completely sure if the decimals can be ignored

## Warnings
There are quite a few warnings, I will put the ones that repeated the most.
```bash
nvcc warning : The 'compute_35', 'compute_37', 'compute_50', 'sm_35', 'sm_37' and 'sm_50' architectures are deprecated, and may be removed in a future release (Use -Wno-deprecated-gpu-targets to suppress warning).
```
```bash
/usr/local/cuda/include/thrust/detail/config/cpp_dialect.h:104:13: warning: Thrust requires C++14. Please pass -std=c++14 to your compiler. Define THRUST_IGNORE_DEPRECATED_CPP_DIALECT to suppress this message.
  104 |   THRUST_COMPILER_DEPRECATION(C++14, pass -std=c++14 to your compiler);
```
```bash
/usr/include/glm/detail/type_vec2.hpp(94): warning: __device__ annotation is ignored on a function("vec") that is explicitly defaulted on its first declaration
```
```bash
/usr/include/glm/detail/type_vec2.hpp(94): warning: __host__ annotation is ignored on a function("vec") that is explicitly defaulted on its first declaration
```
```bash
home/alejandro/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/kernels/kernel_Octree.h(149): warning: argument is incompatible with corresponding format string conversion
          detected during:
            instantiation of "void gpu_voxels::NTree::kernel_print<branching_factor,level_count,InnerNode,LeafNode>(InnerNode *, InnerNode *, InnerNode *) [with branching_factor=8UL, level_count=15UL, InnerNode=gpu_voxels::NTree::Environment::InnerNodeProb, LeafNode=gpu_voxels::NTree::Environment::LeafNodeProb]" 
/home/alejandro/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/NTree.hpp(695): here
            instantiation of "void gpu_voxels::NTree::NTree<branching_factor, level_count, InnerNode, LeafNode>::print() [with branching_factor=8UL, level_count=15UL, InnerNode=gpu_voxels::NTree::Environment::InnerNodeProb, LeafNode=gpu_voxels::NTree::Environment::LeafNodeProb]" 
/home/alejandro/gpu-voxels/packages/gpu_voxels/src/gpu_voxels/octree/Octree.cu(63): here
```
```bash
In file included from /usr/include/string.h:495,
                 from /usr/include/c++/9/cstring:42,
                 from /home/alejandro/gpu-voxels/packages/icl_core/src/icl_core_logging/LogOutputStream.cpp:15:
In function ‘char* strncpy(char*, const char*, size_t)’,
    inlined from ‘icl_core::logging::LogOutputStream::LogMessage::LogMessage(const icl_core::TimeStamp&, icl_core::logging::LogLevel, const char*, const char*, size_t, const char*, const char*, const char*, const char*)’ at /home/alejandro/gpu-voxels/packages/icl_core/src/icl_core_logging/LogOutputStream.cpp:659:15:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:34: warning: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ specified bound 257 equals destination size [-Wstringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
```

The new version of GCC now shows more warnnings and for experience some of them can be real bugs in the code, so is worth looking

Cheers

AlejoDiaz49, 2020-12-28:
Here I leave the Dockerfile to easily have this environment and reproduce the same errors and warnings.
```dockerfile
FROM nvidia/cuda:11.0-devel-ubuntu20.04

WORKDIR /home/

# Basic Tools
RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes apt-utils=2.0.2ubuntu0.2 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes sudo=1.8.31-1ubuntu1.1 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes ca-certificates=20201027ubuntu0.20.04.1 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes g++=4:9.3.0-1ubuntu2 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes gcc=4:9.3.0-1ubuntu2 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes git=1:2.25.1-1ubuntu3 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes make=4.2.1-1.2 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes cmake=3.16.3-1ubuntu1 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes vim=2:8.1.2269-1ubuntu5 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

# Libraries & Dependencies
RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes libboost-all-dev=1.71.0.0ubuntu2 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN git clone https://gitlab.com/libeigen/eigen.git && \
    cd eigen && \
    git checkout 3.3.9 && \
    mkdir build && \
    cd build && \
    cmake ../ && \
    make && \
    make install && \
    cd ../../ && \
    rm -rf eigen/

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes libflann-dev=1.9.1+dfsg-9build1 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

ENV TZ=Europe
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone && \
    apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes libvtk7-dev=7.1.1+dfsg2-2ubuntu1 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes libtinyxml-dev=2.6.2-4build1 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes freeglut3-dev=2.8.1-3 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes libglm-dev=0.9.9.7+ds-1 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends --assume-yes libglew-dev=2.1.0-4 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

RUN git clone https://github.com/PointCloudLibrary/pcl.git && \
    cd pcl/ && \
    git checkout pcl-1.9.1 && \
    mkdir build && \
    cd build && \
    cmake ../ -DBUILD_examples=OFF && \
    make -l 2.5 && \
    make install && \
    cd ../../ && \
    rm -rf pcl/

# Create non root user in the sudo group.
SHELL ["/bin/bash", "-o", "pipefail", "-c"]
RUN useradd --create-home --shell /bin/bash --gid sudo gpuvoxels
RUN echo 'gpuvoxels:gpuvoxels' | chpasswd
USER gpuvoxels
WORKDIR /home/gpuvoxels
```

Squelsh, 2020-12-29:
Thanks a lot, @AlejoDiaz49 !!