fzi-forschungszentrum-informatik/gpu-voxels #123: Problems with linking example

crlz182, 2021-03-25:
Hello,

I have compiled gpu-voxels successfully and now I am trying to execute the linking example (example_how to_link). I am following the README but get runtime errors. I am new to C++ so that's most likely a trivial problem. Here is what I did:

Installing gpu-voxels
` cmake -D CMAKE_C_COMPILER=gcc-6  -D CMAKE_CXX_COMPILER=g++-6 -DCMAKE_INSTALL_PREFIX=/home/carl/gvl/ ..`
`make && make install`

in the build folder of the example:

`(base) carl@carl-pc:~/Dokumente/gpu-voxels/example_how_to_link/build$ cmake .. -DCMAKE_PREFIX_PATH=/home/carl/gvl -DCMAKE_C_COMPILER=gcc-6 -DCMAKE_CXX_COMPILER=g++-6
-- The C compiler identification is GNU 6.4.0
-- The CXX compiler identification is GNU 6.4.0
-- Check for working C compiler: /usr/bin/gcc-6
-- Check for working C compiler: /usr/bin/gcc-6 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/g++-6
-- Check for working CXX compiler: /usr/bin/g++-6 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr (found version "9.1") 
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
-- GVL include dirs are: /home/carl/gvl/include
-- Configuring done
-- Generating done
-- Build files have been written to: /home/carl/Dokumente/gpu-voxels/example_how_to_link/build
`

Building the example with `make`

`./gvl_linkage_test 
./gvl_linkage_test: error while loading shared libraries: libicl_core_config.so: cannot open shared object file: No such file or directory
`

Could you give me a hit what went wrong here?

cjue, 2021-03-25:
Hi, first off: do you find several files matching the `libicl_core*.so` and `libgpu_voxels*.so` in the `lib` subfolder of your install location?

If not, CMake was probably missing some dependencies for gpu_voxels and did only a partial build.

crlz182, 2021-03-25:
> Hi, first off: do you find several files matching the `libicl_core*.so` and `libgpu_voxels*.so` patterns in the `lib` subfolder of your install location?
> 
> If not, CMake was probably missing some dependencies for gpu_voxels and did only a partial build.

Thanks for your quick answer! Yes, I can find these files there:

`(base) carl@carl-pc:~/gvl/lib$ ls
libgpu_voxels_core.so                    libgpu_voxels_voxellist.so
libgpu_voxels_dh_robot.so                libgpu_voxels_voxelmap.so
libgpu_voxels_distance_map_converter.so  libgpu_voxels_voxelmap_tests.so
libgpu_voxels_helpers.so                 libgpu_voxels_voxel.so
libgpu_voxels_logging.so                 libicl_core_config.so
libgpu_voxels_octree.so                  libicl_core_dispatch.so
libgpu_voxels_primitive_array.so         libicl_core_logging.so
libgpu_voxels.so                         libicl_core_performance_monitor.so
libgpu_voxels_urdf_robot.so              libicl_core_plugin.so
libgpu_voxels_vis_interface.so           libicl_core.so
libgpu_voxels_visualization_core.so      libicl_core_thread.so
`
I am also able to run the example programs such as the OMPL planning example.

cjue, 2021-03-25:
So the GPU-Voxels build appears to be complete, and the linking-example builds as well.

Have you tried to add the installed lib location to your library search path? This should work:

`LD_LIBRARY_PATH=/home/carl/gvl/lib:$LD_LIBRARY_PATH ./gvl_linkage_test`

crlz182, 2021-03-25:
> So the GPU-Voxels build appears to be complete, and the linking-example builds as well.
> 
> Have you tried to add the installed lib location to your library search path? This should work:
> 
> `LD_LIBRARY_PATH=/home/carl/gvl/lib:$LD_LIBRARY_PATH ./gvl_linkage_test`
> 
> Sidenote: with this syntax the LD_LIBRARY_PATH is only changed for the command in the same line, which in this case is `./gvl_linkage_test`

I can execute it now. However, I get 0 collisions instead of the 8000 mentioned in the README. But I am able to link now and the problem is solved. Thanks a lot again. :)

closed 2021-03-26 by crlz182