fzi-forschungszentrum-informatik/gpu-voxels #124: Changing a URDF robot pose

mfinean, 2021-04-01:
I'm currently trying to sweep the volume of a robot (loaded from URDF) into a map. SO far I've mostly followed the RobotVsEnvironment example and replaced the DH model with a URDF robot. Unfortunately, the URDF robot doesn't have simple joints which I can use to manipulate the base pose. How can I manually change the pose of the robot each iteration that I want to insert it into the map? Many thanks

felixvd, 2021-07-05:
I did not test this, but [this JointValueMap](https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/examples/RobotVsEnvironment.cpp#L162) should work for the URDF robot as well, since [setConfiguration](https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/gpu_voxels/robot/urdf_robot/robot.cpp#L208) is defined in the Robot parent class. I assume you'd use that to change the robot pose.

cjue, 2021-07-05:
Exactly, you can set the joints of a URDF robot like this:

https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/d77afbb557cd400ed4c81504444cd3db612d67e7/packages/gpu_voxels/src/examples/URDF_Loader_ROS_listener.cpp#L69

Thanks for your input, @felixvd!

closed 2021-07-05 by cjue