fzi-forschungszentrum-informatik/gpu-voxels #126: Visualization failed, possibly caused by corrupted shared memory files!

crlz182, 2021-07-14:
I am having problems visualizing my voxel maps. I am using the ompl planning example as a base line and added some code on top of that. The visualizer crashes every second time with the following error message. Could this be caused by how I exit my program? I am often forced to use ctrl+c to kill my ROS node. Which role does the `gvl.reset()` command play in that context? After restarting everything it works again but I would like to avoid that workaround in the future. Thanks a lot in advance!

`<2021-07-14 09:28:04.509> Visualization(Info)::main: Starting the gpu_voxels Visualizer.
<2021-07-14 09:28:04.509> Visualization(Info) Visualizer::initializeVisualizer: Trying to open the Visualizer shared memory segment created by a process using VisProvider.
<2021-07-14 09:28:04.636> Visualization(Info) Visualizer::initGL: Using GPU Device 0: "GeForce GTX 1660 Ti" with compute capability 7.5
<2021-07-14 09:28:04.676> Visualization(Info) Visualizer::initGL: Using OpenGL Version: 4.6.0 NVIDIA 460.84
<2021-07-14 09:28:04.677> Visualization(Info)::main: Number of voxel maps that will be drawn: 5
<2021-07-14 09:28:04.677> Visualization(Error)::registerVoxelmapFromSharedMemory: Couldn't find mem_segment voxel_map_handler_dev_pointer_0
<2021-07-14 09:28:04.677> Visualization(Error)::registerVoxelmapFromSharedMemory: Couldn't find mem_segment voxel_map_handler_dev_pointer_1
<2021-07-14 09:28:04.677> Visualization(Error)::registerVoxelmapFromSharedMemory: Couldn't find mem_segment voxel_map_handler_dev_pointer_2
<2021-07-14 09:28:04.677> Visualization(Error)::registerVoxelmapFromSharedMemory: Couldn't find mem_segment voxel_map_handler_dev_pointer_3
<2021-07-14 09:28:04.677> Visualization(Info)::registerVoxelmapFromSharedMemory: Providing a PROBAB_VOXELMAP called "myEnvironmentMap" with side_length 0.010000 and dimension [400, 400, 400]
<2021-07-14 09:28:04.678> Visualization(Warning) XMLInterpreter::getDataContext: Occupancy_threshold of /myEnvironmentMap is zero.
<2021-07-14 09:28:04.678> Visualization(Info)::main: Number of voxel lists that will be drawn: 1
<2021-07-14 09:28:04.678> Visualization(Info)::registerVoxellistFromSharedMemory: Providing a voxellist called "mySolutionMap"
<2021-07-14 09:28:04.678> Visualization(Warning) XMLInterpreter::getDataContext: Occupancy_threshold of /mySolutionMap is zero.
<2021-07-14 09:28:04.679> Visualization(Warning) XMLInterpreter::getDataContext: Occupancy_threshold of /voxellist_0 is zero.
<2021-07-14 09:28:04.679> Visualization(Warning) Visualizer::registerVoxelList: No context found for voxel list mySolutionMap. Using the default context.
<2021-07-14 09:28:04.679> Visualization(Info)::main: Number of octrees that will be drawn: 0
<2021-07-14 09:28:04.679> Visualization(Info)::main: Number of Primitive Arrays that will be drawn: 0
<2021-07-14 09:28:04.749> Visualization(Error)::main: Visualization failed, possibly caused by corrupted shared memory files!
<2021-07-14 09:28:04.749> Visualization(Error)::main: Error message: 'parallel_for failed: invalid resource handle'. Removing shared memory files in /dev/shm/...
<2021-07-14 09:28:04.749> Visualization(Error)::main: Cleaned up shared memory. Please try to restart your GPU-Voxels processes and then the visualizer.
<2021-07-14 09:28:04.749> Visualization(Info)::main: Exiting..
Speicherzugriffsfehler (Speicherabzug geschrieben)
`

cjue, 2021-07-15:
Hi, is it possible that the way shut down your program with Ctrl-C the gvl.reset() never runs?

The examples have Signal handlers for shutdown signals that try to make sure the GpuVoxel destructor is run, which should clear all leftover shared memory files.

If gvl.reset() does run when you shutdown the program and there are still Voxelmap/Voxellist/etc. files left over in /dev/shm/ I can have a closer look.

crlz182, 2021-07-15:
Thank you for your quick reply! In the gvl_ompl_planning example the `reset()` function is called in the destructor:

`GvlOmplPlannerHelper::~GvlOmplPlannerHelper()
{
    gvl.reset(); // Not even required, as we use smart pointers.
}`

This, however, does not clean up the files. I managed to solve the problem by explicitely calling the destructor from the main program (gvl_ompl_planner.cpp) in the way you suggested:

`void ctrlchandler(int)
{
  my_class_ptr->~GvlOmplPlannerHelper();
  exit(EXIT_SUCCESS);
}
void killhandler(int)
{
  my_class_ptr->~GvlOmplPlannerHelper();
  exit(EXIT_SUCCESS);
}

int main(int argc, char **argv)
{

    signal(SIGINT, ctrlchandler);
    signal(SIGTERM, killhandler);
...}`

closed 2021-07-15 by crlz182