fzi-forschungszentrum-informatik/gpu-voxels #128: How to convert pointcloud to octree

wing0716, 2022-02-25:
Hi, thank you very much for your research work, I noticed that you use the octree data structure to identify the environment map, but how to convert pointcloud to octree? I don't know which function to use and related API

cjue, 2022-02-25:
The octree functionality of gpu-voxels has severe issues on newer CUDA hardware, therefore I would advise against using them at the moment.

Examples for the usage of octrees can be found here:
* https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/examples/Sandbox.cpp
* https://github.com/fzi-forschungszentrum-informatik/gpu-voxels/blob/master/packages/gpu_voxels/src/examples/SweptVolumeVsEnvironment.cpp

closed 2022-02-25 by cjue

wing0716, 2022-02-25:
Thank you for your quick reply, I'm using GTX3060 and CUDA11.2 platform now and do have some issues when I run the sandbox, robot_vs_environment and swept_volume_vs_environment examples.
So my question is, on newer CUDA hardware which data structure should I use for the environment map pointcloud in collision detection, is it voxellists or voxelmap? thank you very much!

cjue, 2022-02-25:
If you just want to detect collision between single point clouds and a dynamic object like a robot the fastest solution is usually using VoxelLists for both.

wing0716, 2022-02-25:
Thanks very much!

wing0716, 2022-02-28:
Sorry I have  problems again.
I used two methods to convert pcd files into VoxelLists. The first method is to use the insertPointCloudFromFile function directly. The second method is to convert the pcd file into a PointCloud data structure first, and then use the insertPointCloud function.
But the running time of these two methods is very different. The total time of the two steps in the second method is much smaller than that of the first direct method, almost an order of magnitude. 
Is there something wrong with my test or it should be like this? I am  confused.